d-ronin/dRonin #1143: Flight: RC input is required even if there is no matching output

DTFUHF, 2016-07-25:
* Seen in version (release version or githash):  Samsara release
* On flight controller (if applicable): DTFc

Issue details:

On setting up RC inputs, a Critical warning is thrown if one of Roll, Pitch, Yaw, Throttle is not configured. There are airframe types that don't have yaw outputs (flying wing, some fixed wing) and of course some with no throttle output (gliders). If these outputs are not configured in the Vehicle page for the aircraft, there should be no warning when the input is not configured.

lolorc, 2016-09-11:
probably related, it also wrongly warns with "no rc input" when you forget to define your arming switch input

mlyle, 2018-04-17:
Also related, there's legal aircraft configurations with output channels missing.