d-ronin/dRonin #1160: Invalid version error when replaying log files

muratozkan, 2016-07-30:
* Seen in version (release version or githash): 20160720.1
* On flight controller (if applicable): Openpilot Revo (using firmware for 20160720.1)

Issue details:

I enabled the logging module and downloaded the flight log file after a short flight. When loaded in GCS, GCS shows the following error:

`The log file was made with branch Release-20160409.2:b04a62e2 20160428, UAVO hash 02cf59ebb14892dc3e52c39c48f32b675361397a. GCS will attempt to play the file.`

[image]

Then these messages appear: 

 - `GCS cannot find the separation byte. GCS will attempt to play the file`
 - `Timestamps are not sequential. Playback may have unexpected behavior`

I am using GCS on OSX Yosemite. Log file is attached.

Regards,

[dRonin-2016-07-30_12-11-03.drlog.zip](https://github.com/d-ronin/dRonin/files/391801/dRonin-2016-07-30_12-11-03.drlog.zip)

mlyle, 2016-07-30:
Unfortunately, the GCS log playback is not the best.   It's not really useful, but also flight firmware and logs recorded with GCS have slightly different formats, and I don't think the GCS one understands flight logs well.

If you're familiar with Python there's a new log viewing tool you can play with-- not quite finished but still useful.

Commands to get it on OSX should be:

```
pip install dronin[all]
dronin-logview
```

mlyle, 2016-07-30:
@muratozkan PS if you try this please let us know how it goes-- whether good or bad.  Feedback on the log viewer tools would be really helpful, as I hope to ship them relatively soon.

muratozkan, 2016-07-31:
Thanks for the reply @mlyle.

I've prepared a `virtualenv` with Python3 (3.5.1) and tried installing dronin. First, dronin-logview complained aboud `lxml` package not being present. Maybe you could include it as a dependency? 

I couldn't get PyQt4 or PySide installed, which are required by `pyqtgraph`. PyQt4 has no packages in pypi and PySide only supports Python up to 3.4 Is there a guide for setting up this package?

mlyle, 2016-07-31:
Woops--- both on lxml (will fix) and pyqt4.  Pyqt4 is tricky because it includes the qt4 libraries and so doesn't fit into pypi so well.  

Personally I just use anaconda which includes everything (pyqt4 etc) except pyqtgraph.

I am also working on a pyinstaller-package of the log viewer, which works but is a little clunky (this is how we will probably officially release it)

muratozkan, 2016-08-15:
Is there any WIP commit or branch regarding the installer? I'd be glad to help with that.

mlyle, 2016-08-15:
Unfortunately I have not committed the spec file.. it's mostly makefile and jenkins-ci stuff to do now.

Current code in github works with pyqt5 which has wheels for python 3.

There is an interim package Windows at http://jar.lyle.org/~mlyle/logview.zip , too.

I've been working on flyingpi quite a bit but this is on my list to get onto our build server for the next release.

mlyle, 2016-09-08:
Dup #1310

closed 2016-09-08 by mlyle