d-ronin/dRonin #1166: Brain (OG & RE1) reset in-flight and after crash

ufoDziner, 2016-08-02:
* Seen in version (release version or githash): bfe4b82 & 5f25d7d
* On flight controller (if applicable): OG Brian, and RE1

Issue details:
Had 1 in-flight reset, and 2 resets after a crash. I don't believe that the crash was caused by the reset, but it's possible that they were. The in-flight reset was with an OG brain using bfe4b82. The 2 crash resets were with an RE1 was using 5f25d7d. "INDY WDOG" was displayed on the screen as the reboot reason each time. These reboots happened somewhere mid-flight (I don't have an accurate time), but powered on for at least 5 min.

mluessi, 2016-08-02:
Both versions include the receiver changes and also #1107 , correct? What type of receiver are you using and is there a way for you to reproduce it on the ground?

ufoDziner, 2016-08-02:
The build on the OG Brain is a Jenkins build from #1135, so I don't think it has all of the changes you're referring to. The RE1 build is a nightly from a couple of days ago, which should. I don't know of a way to reproduce it on th ground.

I flew 2 packs yesterday on the RE1 with no issues, but the total time powered up was <10min. The flights on the other day were powered on for much longer. Not sure if it is a factor, as I don't know how long they were powered up prior to the reset.

ufoDziner, 2016-08-02:
The build on the OG Brain is a Jenkins build from #1135, so I don't think it has all of the changes you're referring to. The RE1 build is a nightly from a couple of days ago, which should. I don't know of a way to reproduce it on th ground.I flew 2 packs yesterday on the RE1 with no issues, but the total time powered up was <10min. The flights the other day were powered on for much longer. Not sure if it is a factor, as I don't know how long they were powered up prior to the reset.                                                                                                                                                                                                                 From: Martin LuessiSent: Tuesday, August 2, 2016 05:28To: d-ronin/dRoninReply To: d-ronin/dRoninCc: Wes Warner; AuthorSubject: Re: [d-ronin/dRonin] Brain (OG & RE1) reset in-flight and after crash (#1166)Both versions include the receiver changes and also #1107 , correct? What type of receiver are you using and is there a way for you to reproduce it on the ground?

ufoDziner, 2016-08-02:
Not sure if it matters, but one receiver is sbus and the other is srxl.

mlyle, 2016-08-02:
Just one more data point-- were the new crashes with logging?

ufoDziner, 2016-08-02:
‎OG Brain has no logging. RE1 has logging enabled, but it doesn't appear to be functioning properly.                                                                                                                                                                                                                 From: Michael LyleSent: Tuesday, August 2, 2016 08:33To: d-ronin/dRoninReply To: d-ronin/dRoninCc: Wes Warner; AuthorSubject: Re: [d-ronin/dRonin] Brain (OG & RE1) reset in-flight and after crash (#1166)Just one more data point-- were the new crashes with logging?

mluessi, 2016-08-04:
@ufoDziner if possible, can you test if this also happens with the stable release? I have been flying the release (or versions very close to it) and haven't experienced this problem. It's possible that the RX system changes are causing this, but the code looks good, so it's a bit of a mystery to me.

ufoDziner, 2016-08-04:
@mluessi Tested with stable just now.  First flip I tried LoS, the FC rebooted mid flip.  Boot message was "INDY WDOG".  I swapped out PDBs last night to ensure that there was no voltage sag contributing to the issue.  Immediately following the flight with the RE1, I put the Release on a OG brain equipped quad with the same power system (except the PDB).  I could not get it to reset in flight.  The only in-flight resets with the OG brain have been fpv.

ufoDziner, 2016-08-24:
Have had zero issues with the a new RE1, which leads me to believe that board was the issue.  Had no issues with a different OG brain as well.  Seems probable that it was also hardware related.

closed 2016-08-24 by ufoDziner

mlyle, 2016-08-24:
@ufoDziner for clarity, does this include next?

ufoDziner, 2016-08-25:
Yes.  Flew next this past weekend and no reboots on the OG brain.  Did not get to fpv the RE1, but flew the same LoS figues that were crashing it before with no issues.