d-ronin/dRonin #1179: #1066 broke Windows build

mlyle, 2016-08-06:
On MSVC:

```
moc_picocsettings.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const HwShared::staticMetaObject" (?staticMetaObject@HwShared@@2UQMetaObject@@B)
modulesettings.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const HwShared::staticMetaObject" (?staticMetaObject@HwShared@@2UQMetaObject@@B)
picocsettings.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const HwShared::staticMetaObject" (?staticMetaObject@HwShared@@2UQMetaObject@@B)
uavobjectsinit.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const HwShared::staticMetaObject" (?staticMetaObject@HwShared@@2UQMetaObject@@B)
moc_modulesettings.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const HwShared::staticMetaObject" (?staticMetaObject@HwShared@@2UQMetaObject@@B)
```

According to @jihlein, on mingw:

```
warning: 'HwSparky::NAME' 
                 redeclared without dllimport attribute:
```

jihlein, 2016-08-06:
Looking thru the build output, I think it's all targets.  Here's more info if it's helpful, this time Naze.

[image]

tracernz, 2016-08-06:
QT macros (Q_OBJECT?) must expand differently on Windows. Sigh, I guess we'll have to revert. I hate Windows :-(.

closed 2016-08-07 by mlyle