d-ronin/dRonin #1187: Sparky2 and FS ia10 Reciever not working

cam4869, 2016-08-08:
* Seen in version (release version or githash): 2016720.1
* On flight controller (if applicable): Sparky 2

Issue details:
Sparky2 FC does not recognize FlySky ia10 receiver connected to RCV port.
Configured for PPM.
Works fine with TauLabs latest build for Sparky2.

mlyle, 2016-08-08:
What exactly do you mean by "does not recognize"?  You go into input wizard and move the throttle when prompted but nothing happens?  Or something else?

cam4869, 2016-08-08:
"RC Input" system status is Yellow..  says no receiver detected.  And the input wizard does not detectect any stick movements.

mlyle, 2016-08-08:
OK, "RC Input" will be yellow or red until things are configured (this is expected).

Unfortunately none of our developers have the flysky hardware.. but at the same time there's been no change to pios_ppm between Tau Labs & the 20160720.1 release.  A couple of ideas though.

Are you using the same Sparky2 and ia-10 for both Tau & dR?  I have seen some reports of iA-10s which PPM doesn't work for..

There is one thing we should probably errata-- you're not using the buffered outputs 7 & 8 are you?  Unfortunately sparky2 has a hardware bug where the same timer is used for input capture and those two outputs, so we can't simultaneously decode PPM and drive those outputs.  (Tau Labs would handle those for -some- actuator settings but not others-- e.g. not if you set them to 12MHz etc).  

You could try using iBus instead...

cam4869, 2016-08-08:
Thank You for the quick support!   I seem to have fixed it..   I reflashed the dRonin to it again just now after running TauLabs this afternoon.  I also changed the wiring from the Sparky2 to the FS-iA10 to include only the signal wire..  not + and -  power..   I fed a clean 5v from the Power control board.

Seems to be working now.  May have been a problem with the voltage coming from the Sparky2 RCV pins was a little less than 5V  (4.5 to be exact!).

So glad to be back to dRonin as you all are more active than Tau.  ;)

Thanks again for your help.

mlyle, 2016-08-08:
Glad it works for you now.

closed 2016-08-08 by mlyle

cam4869, 2016-08-10:
Well..  not sure whats up but PPM quit working again out of the blue..   But I did switch to iBUS and it works great.  Got 10 channels instead of 8 too!

So I too recommend just use iBUS with the iA10 receiver. 

Thanks for the help!