d-ronin/dRonin #1192: Radio Wizard fails to detect on CC3D

ragd011, 2016-08-10:
* Seen in version (release version or githash): dRonin 2016-07-20.1 Release ("Samsara")
* On flight controller (if applicable): CC3D

Issue details:  After erasing configuration of FC the Radio Setup Wizard fails to detect the throttle when using PPM on pin 8 for the receiver.  Resolution was to flash back to previous release dRonin 2016-07-20 Release ("Samsara") and it worked on the first attempt.

I flashed to the latest version and erased my settings to try to troubleshoot a tuning issue.  When I was going through the wizards I got to the Radio and it wouldn't detect the throttle stick moving.  I tried a few different things including moving the radio (9XR) across the room from the rx (D8R-II flashed with CPPM) but nothing worked.  I installed the previous release and flashed the FC back and setup ran on the first attempt.

System is Win 10 X64 with latest updates.  Tried it on a desktop win 10 and Laptop Win 10 with same results.

Thanks, Rob.

joeynuggetz, 2016-08-11:
I had some issues initially too with the wizard. It finally worked but then the last window that allows you to move the sticks and switches to make sure the actions on screen mirror reality failed to work. Unfortunately I do not have a good repeatable process to reproduce but I experienced some "wonkyness"

Revo with X4R/Taranis

mlyle, 2016-09-09:
Dup #1317.

closed 2016-09-09 by mlyle