d-ronin/dRonin #1193: GCS scaling issues at 3000x2000 resolution

joeynuggetz, 2016-08-11:
* Seen in version (release version or githash): Release-20160720.1: 20160723-74b7422c
* On flight controller (if applicable): Revo

Issue details:
Running GCS on Windows 10 Surface Book at native resolution of 3000x2000 exhibits some scaling issues. Also, the dialog boxes during automatic firmware upgrade process need to be resized in order to read the text properly. I don't have a screenshot of that however...

- Channel dropdown boxes only displays 6 characters 
- Font size for section headers appear much too small when compared to the rest of the text on the page. For example, the words "Calibration and Configuration Options" sort of seem unusually small.q

[image]

tracernz, 2016-08-11:
Note: Don't see these issues on OSX with Hi-DPI.

mlyle, 2017-03-12:
Dup #533

closed 2017-03-12 by mlyle