d-ronin/dRonin #1208: RE1 LEDs randomly wrong color

ufoDziner, 2016-08-17:
* Seen in version (release version or githash): Release 20170720.1
* On flight controller (if applicable): BrainFPV RE1

Issue details:
LEDs will be multiple colors on powerup rather than the selected color.

Example:
Color set to yellow for a bank of 8 LEDs.  On boot the first 2 were blue, 3rd magenta, next 5 yellow (I think that was the mix/order).

mluessi, 2016-08-17:
I assume this can happen due to power sequencing when connecting the battery. I.e. the LEDs don't have full power, bc they need 5V and the RE1 starts earlier (3.3V), so if the power ramp is not very fast, the LEDs may not function properly. We should add a actual LED module that keeps updating the LEDs, then this shouldn't happen.

mlyle, 2016-09-01:
Curious to know if any BOR fix helps this..

mluessi, 2016-09-23:
@mlyle One way to fix this until we have a more proper RGB LED module could be to start a callback timer on boot so the LEDs get set again after 1s or so after boot. WDYT? Too much of a hack?

AlessioMorale, 2016-09-23:
When working on ws2811 I had a lot of troubles with leds powered with slightly more than 5V. Looks like logic levels are referenced to supply voltage. a simple diode may do the job.

[image]

mluessi, 2016-09-23:
@AlessioMorale this should not be a problem on RE1, it actually uses a FET to generate 5V logic output for the LEDs.

mlyle, 2016-09-23:
@mluessi -- I'm fine with an approach like this-- we have other modules that are callback-only effectively.  Should stay within the target for now.

But I'd also like to see appropriate BOD programming for RE1 as part of this release. 

We're supposed to be functionally complete today-- which we won't make-- but ASAP please.

PR #1382 (2016-09-24): BrainRE1: set BOR, update LEDs periodically
Sets the brown-out threshold and updates the programmable LEDs periodically. 

Status: bench tested, should be good.

Fixes #1208
closed 2016-09-24 by mlyle