d-ronin/dRonin #1216: Safety-Related  Motors spin when connecting battery

ufoDziner, 2016-08-21:
* Seen in version (release version or githash): f1c776c9
* On flight controller (if applicable): DTFc

Issue details:
  While plugging in the battery on a few occasions the 2 back motors spun.  @mlyle suggested testing the "reboot" button in the GCS with the battery connected to see if they spin.  I tried this 3 times, and all 3 times at least one motor spun up, like it was armed then disarmed immediately.

mlyle, 2016-08-21:
@DTFUHF What do you think of this?

tracernz, 2016-08-21:
What kind of setup (e.g. Oneshot125?)? ESC firmware?

ufoDziner, 2016-08-21:
Oneshot125, V-Good 30A with their latest firmware.

tracernz, 2016-08-21:
Thanks, seems likely this isn't DTFc specific @mlyle. Sounds like TauLabs/TauLabs#1821...

mlyle, 2016-08-21:
@tracernz yes, except we had #719 which cured it on sparky2

mlyle, 2016-09-09:
@jihlein pointed out we spend longer in bootloader (500ms) on dtfc even though we have a sense line..  Which should be fixed.

tracernz, 2016-09-09:
vsense is defined too :/ https://github.com/d-ronin/dRonin/blob/next/flight/targets/dtfc/board-info/board_hw_defs.c#L838

DTFUHF, 2016-09-09:
@tracernz @jihlein probably copypaste fail. Changing to 2 (like sparky1) sounds OK to me.

PR #1324 (2016-09-20): firmwareiap: stop PWM before reset (when appropriate)
Fixes #1216, where @ufoDziner has a reliable repro.

Supersedes #720.

Needs to wait for #1323 / build to be fixed before building artifacts.

Conflicts #1318
closed 2016-09-20 by tracernz