d-ronin/dRonin #1220: Vehicle type not saved

ufoDziner, 2016-08-22:
* Seen in version (release version or githash): 20160720.1
* On flight controller (if applicable): Sparky2

Issue details:
  Vehicle type switches back to Multirotor on the Vehicle tab after saving as Fixed wing.  Further investigation shows that only Multirotor & Custom will save, Fixed wing and Helicopter default back to either Multirotor or Custom (whichever was saved last), and Ground sends the FC into a bootloop.

mlyle, 2016-08-22:
SmartSaveButton emits preProcessOperations().. which in turn gathers the data from the UAV widgets... but expects it to be done immediately instead of asynchronously as signals may.  So old/stale UAVO data is stored.

mlyle, 2016-08-22:
#315 possibly related

tracernz, 2016-08-22:
If the signal and slot are in the same thread, the default connection is direct which will cause the slots to all process immediately from the signal handler, before the thread continues.

mlyle, 2016-08-22:
Another possible option is the vehicle-specific pages are configtaskwidgets and so is their parent page..

tracernz, 2016-09-08:
Maybe something to do with this?: https://github.com/d-ronin/dRonin/blob/080150a5b271fcfb3fe36aa1e20f92fc0d956aca/ground/gcs/src/plugins/config/configvehicletypewidget.cpp#L435

mlyle, 2016-09-29:
Encountered a cousin of this.

I upgraded from an old random build on my CC3D.  All of configuration was upgraded, EXCEPT the mixer settings were reset to default.

My motor mapping displayed correctly in GCS but was actually not and was flipping in takeoff until i saved in vehicle setup pane twice.

End users are not likely to encounter this, because the upgrader will know about their githash.. but still this severely sucks.

mlyle, 2017-02-04:
Series of objects sent to other side and save are correct.  However a stale SystemSettings is sent.  Disconnecting and connecting immediately after this sequence yields the stale data.

```
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectwidgetutils/smartsavebutton.cpp:140 void smartSaveButton::processOperation(QPushButton*, bool)] [smartsavebutton.cpp] Sending object to remote end -  "SystemSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectwidgetutils/smartsavebutton.cpp:177 void smartSaveButton::processOperation(QPushButton*, bool)] [smartsavebutton.cpp] Save request for object "SystemSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:117 void UAVObjectUtilManager::saveObjectToFlash(UAVObject*)] "Enqueue object:SystemSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:142 void UAVObjectUtilManager::saveNextObject()] "Send save object request to board SystemSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:157 void UAVObjectUtilManager::saveNextObject()] "[saveObjectToFlash] Moving on to AWAITING_ACK"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:197 void UAVObjectUtilManager::objectPersistenceTransactionCompleted(UAVObject*, bool)] "[saveObjectToFlash] Moving on to AWAITING_COMPLETED"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:275 void UAVObjectUtilManager::objectPersistenceUpdated(UAVObject*)] "[saveObjectToFlash] Object save succeeded"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectwidgetutils/smartsavebutton.cpp:140 void smartSaveButton::processOperation(QPushButton*, bool)] [smartsavebutton.cpp] Sending object to remote end -  "MixerSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectwidgetutils/smartsavebutton.cpp:177 void smartSaveButton::processOperation(QPushButton*, bool)] [smartsavebutton.cpp] Save request for object "MixerSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:117 void UAVObjectUtilManager::saveObjectToFlash(UAVObject*)] "Enqueue object:MixerSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:142 void UAVObjectUtilManager::saveNextObject()] "Send save object request to board MixerSettings"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:157 void UAVObjectUtilManager::saveNextObject()] "[saveObjectToFlash] Moving on to AWAITING_ACK"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:197 void UAVObjectUtilManager::objectPersistenceTransactionCompleted(UAVObject*, bool)] "[saveObjectToFlash] Moving on to AWAITING_COMPLETED"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectutil/uavobjectutilmanager.cpp:275 void UAVObjectUtilManager::objectPersistenceUpdated(UAVObject*)] "[saveObjectToFlash] Object save succeeded"
09:35:01[debug][/home/mlyle/dRonin/ground/gcs/src/plugins/uavobjectwidgetutils/smartsavebutton.cpp:240 void smartSaveButton::processOperation(QPushButton*, bool)] RESULT "All operations finished successfully" () () ()
```

PR #1549 (2017-02-05): Vehicle setup pane: properly keep configuration
Fixes #1220 

This code is a huge mess and needs a rewrite.

I've banged on this quite a bit but could use additional test feedback due to criticality and how late we are-- both the vehicle setup pane and the setup wizard.
closed 2017-02-05 by tracernz