d-ronin/dRonin #1222: Create FAQ page for website

jhitesma, 2016-08-22:
It seems we're seeing a lot of the same questions repeat in the forum, FB, and IRC. 

I'd be willing to put together a page with concise answers to the most common questions to try and help cut down on some of the support overhead.

A few questions I can think of:

- Things feel sluggish after autotune how can I make it snappier?
- Yaw is slow how can I make it faster?
- I'm getting bad autotune results what should I try next?
- What makes dRonin different from *flight FW's?
- How do I setup Hangtime?

There are more but those are the ones that come to mind for me off the top of my head.  I'm sure there could be a whole section of "*flight does X how do I do that on dRonin" but I haven't used *flight in so long now I'm not sure what would help most there.

jhitesma, 2016-08-22:
Few more I just thought of:

- Should I use Multishot?  If so how do I set it up?
- Should I use PWM, PPM or SBUS for RX connection?
- *flight - How do I change looptime?

mlyle, 2016-08-22:
It has a slight lean from the perfect hover attitude?  (Level configuration, input settings, or manualcontrol off).

tracernz, 2016-08-22:
This should probably go in the docs?

jhitesma, 2016-08-22:
@tracernz - I'm not aware of a way to PR a new page in the docs, I know I can suggest edits to existing pages but don't see any way to propose a new page, I could just be missing it though.  Adding it as a page on the site I can just do a PR on the site repo, if there's a way to propose a new page in the docs I'd love to know about it!

I could add a page to the wiki for it...but...I got the impression we're trying to get away from using the wiki.

And even if it does go in the docs instead of a page on the site we should add a link in the main menu bar of the site to it so it's obvious since so many people don't seem to read the docs.

yamahabest, 2016-08-22:
Docs would be the right place.
Creating a page on the website would be like writing a documentation system, whilst there is already one in place.

The docs link is already there in the main menu bar of the website, perhaps the docs could be highlighted more on the main page.

dustin, 2016-08-22:
"How do I change looptime?":  http://xyproblem.info/

tracernz, 2016-08-23:
@yamahabest Yah, we already have two different doc systems, don't need a third.

@jhitesma I've not worked on the docs before except in the wiki (it's nice and easy). @mlyle knows how/has access to the readthedocs stuff.

tracernz, 2016-08-23:
P.S. We already have a FAQ https://github.com/d-ronin/dRonin/wiki/@FAQs

mlyle, 2016-08-23:
If someone comes up with a draft of a document, I'm glad to provide additional content, edit, and get into dronin.readme.io.

tracernz, 2016-08-23:
We should put
>   http://xyproblem.info/

at the top.

mlyle, 2016-08-23:
"Spin while armed" / failsafe questions seem to be trending, too

jhitesma, 2016-08-23:
I'm willing to do some editing and cleanup.  But the question is do we want to update the wiki version or do we want to add a new version to the readme.io docs?

Either way I think the main site needs to be updated with a direct obvious link since a simple FAQ page is less intimidating and therefore more likely to get read than full documentation.   I can do a PR to update the website but need to know if I should put effort into building something new for the readme.io site or just update the wiki which I thought we were trying to transition away from.

mlyle, 2016-10-02:
https://dronin.readme.io/v201610-xx/docs/frequently-asked-questions

closed 2016-10-02 by mlyle