d-ronin/dRonin #1223: Change default pitch stick interpretation

dustin, 2016-08-23:
By default, moving the pitch stick forward pitches up.  This violates all the newcomer's expectations until they eventually learn to just click the checkbox to reverse the axis.

There was a discussion somewhere in Tau Labs about why it was done this way, but it's surprising.

I know people aren't excited about changing the wizard, but possible improvements:

1. Ask for specific extents (move sticks to bottom left; move sticks to top right)
2. Just check the box by default

(2 has the advantage of being backwards compatible with people's brains, but might still lead people to wonder why it is the way it is)

tracernz, 2016-08-23:
Huh? The issue is that a lot of radio systems have pitch backwards by default. Many also have other axes backwards by default and it depends entirely on the users radio configuration.

mlyle, 2016-08-23:
I really don't wanna tinker with input wizard.. it's haunted

closed 2016-09-09 by mlyle