d-ronin/dRonin #1224: make package_installer error

ewingate, 2016-08-23:
I'm compiling on a Windows 10 machine.  Make all goes well, but when I try and do a make package_installer it fails when trying to gather GCS debug symbols.  Here's a copy/paste from the bash:

ground_package starting
Gathering GCS debug symbols
No object files found!
makefile:92: recipe for target 'package_ground_symbols' failed
mingw32-make[1]: *** [package_ground_symbols] Error 1
makefile:1067: recipe for target 'package_installer' failed
mingw32-make: *** [package_installer] Error 2

tracernz, 2016-08-23:
Ah, only works for MSVC builds (which we have switched to on our build system). I guess I should fix that so MinGW packaging works (we can't use debug symbols from MinGW builds on Windows for our crash reporting system). Note: we are probably moving away from MinGW once MSVC is proven.

ewingate, 2016-08-23:
So should I give up on making the installer?

tracernz, 2016-08-23:
I'll have a look at this tonight. Should be easy to make it work. :smile:

tracernz, 2016-09-11:
MSVC environment setup documentation WIP instead. Not fixing this.

closed 2016-09-11 by tracernz