d-ronin/dRonin #1225: Android GCS build Error - Windows

ewingate, 2016-08-23:
The make file for the Android GCS doesn't work with Windows.  I've found the following changes help a bit:

ANDROID_SDK_URL := http://dl.google.com/android/android-sdk_r23.0.2-windows.zip
**this link does not have the Build-Tools directory**

#$(V1) wget --no-check-certificate -N -P "$(DL_DIR)" "$(ANDROID_SDK_URL)" 
**need to remove "--no-check-certificate"**

#$(V1) tar -C $(TOOLS_DIR) -xf "$(DL_DIR)/$(ANDROID_SDK_FILE)"
**should be $(V1) unzip -q -d $(TOOLS_DIR) "$(DL_DIR)/$(ANDROID_SDK_FILE)"**

ANDROID_DX  := $(ANDROID_SDK_DIR)/build-tools/22.0.0/dx
**needs to be ANDROID_DX  := $(ANDROID_SDK_DIR)/build-tools/24.0.1/dx.bat, or whatever build tools version you manage to find, the .bat extension is the error**

With these changes it still won't compile, I get the following error:

**ln: failed to create symbolic link 'C:/Users/Office/dRonin/build/uavo-collections/srctree/uavo-xml' -> 'C:/Users/Office/dRonin': File name too long
makefile:445: recipe for target 'C:/Users/Office/dRonin/build/uavo-collections/srctree/uavo-xml' failed
mingw32-make: *** [C:/Users/Office/dRonin/build/uavo-collections/srctree/uavo-xml] Error 1**

tracernz, 2016-08-23:
Thanks for the report. We have moved to using curl on Windows for the Qt and arm-none-eabi-gcc SDKs (msys-git doesn't come with wget by default). We should do this with the Android SDK too.

ewingate, 2016-08-23:
The Linux Android SDK link doesn't have the Build-Tools directory either.

ewingate, 2016-10-19:
These issues are still in the Quixote branch.  I installed WGET for Windows (GNU version) and that got past the wget issue, but the tar and dx.bat issues are still there.

mlyle, 2017-03-07:
@ewingate -- would you like to open a PR to correct this, since you said you've gotten a successful build?  Windows AGCS build support is not likely to be at the top of other developers' priority lists.

ewingate, 2017-03-07:
Ok, will do.  I've switched over to Linux for dRonin builds, but I can set up the build environment for Windows and get it going again.

mlyle, 2017-03-07:
@ewingate If this isn't something you're encountering anymore, I'm glad to close the bug WONTFIX instead ;)

ewingate, 2017-03-07:
That's definitely a more efficient way to deal with it.

closed 2017-03-07 by mlyle