d-ronin/dRonin #1227: flyingpi: easy to lose configuration data

mlyle, 2016-08-24:
* Seen in version (release version or githash): 51bfc90ea
* On flight controller (if applicable): flyingpi

Issue details:
Haven't tracked this down yet.  Have lost configuration objects a few times.  My theories are:

1. We don't commit things to "flash" robustly, and it's possible when I unceremoniously drop power on the pi filesystem data is getting lost.
2. This is TCP telemetry-- could be a bug.
3. Could be a bug with GCS object saving state machine.
4. Could be a bug with uavo browser.

mlyle, 2016-09-01:
OK.  I still believe there's probably a problem, but I just realized that I was running the program from two different directories / two different theflash.bin's.

mhv-shared, 2016-10-24:
I think this happened to me - I flew an autotune without GCS connected, but when I landed and connected the prompt to apply did not come up. I restarted the simulator (without powering down the pi) and the prompt came up, but my input settings had been lost.

mlyle, 2017-06-02:
The flush operations introduced fixed this.

closed 2017-06-02 by mlyle