d-ronin/dRonin #1230: Simulation altitude goes NaN

dustin, 2016-08-25:
* Seen in version (release version or githash): Release-20160720-260-g51bfc90
* On flight controller (if applicable): simulator

dustin, 2016-08-25:
I tried to find a pattern that led to this, but pretty much anything between wiping state and trying to fly led to this:

[image]

I also tried going back to an older version (specifically, the oldest one with the current QT requirement).  It did eventually go NaN on me, but was kind of useless before that, with seemingly random numbers going into attitude actual when armed.

mlyle, 2017-01-13:
@dustin -- at some point we need to work together to run this down, as I still don't have a repro (have seen it once on flyingpi, but.)

mlyle, 2017-03-10:
I believe that I understand this now.  It's racy.  Relates to task switching in a signal handler--- you can't really legitimately do that anymore.  THe major failure here is FP state getting restored by the kernel on sigreturn; if there's multiple task switches in the meantime bad things happen.

PR #1649 (2017-03-25): Big internal refactor of sim / freertos related things
Open for transparency / "dibs":

* [x] Move sim to direct posix threading / scheduling primitives
* [x] Move pipxtreme from freertos to ChibiOS
* [x] Remove freertos
* [x] Begin cleanup of freertos-related code
* [x] ~~Try to identify places that should have the chibios epilogue (uart? usb?) to trigger resched~~ defer to #1650 
* [x] ~~Anomaly: Some tasks on pi seem to not be realtime-class?~~
* [x] Need to somehow ensure precedence of Pi command line args
* [x] Anomaly: make sure pipx still works good

Fixes #1230 
Fixes #169
closed 2017-03-25 by tracernz