d-ronin/dRonin #1234: Motors keep running on Failsafe if "Motors spin at neutral is enabled"

mluessi, 2016-08-25:
* Seen in version (release version or githash): 2016-07-20.1
* On flight controller (if applicable): BrainRE1 (likely affects all targets)

Issue details: On Tx failsafe, the motors keep running at neutral throttle until the arming timeout has expired. This is unsafe, as the default is 30s and the motors / ESCs will likely burn up in a crash. Setting the timeout to 0 make the motors run indefinitely. 

IMO this is a bug. The motors should stop immediately (or after a much shorter period than 30s) on failsafe.

mlyle, 2016-08-25:
#691

mluessi, 2016-08-25:
From discussion on IRC: This is already fixed in the code using ManualControlSettings.InavlidRxArmedTiemout, which is disabled by default but can be set to e.g. 2s to disarm after no valid rx signal is detected. We need to add this to the UI and the MR wizard should set this to 2s or so.

mluessi, 2016-08-25:
superseded by #1235

closed 2016-08-25 by mluessi

tracernz, 2016-08-25:
@mluessi Yes, my position was that 30 s is way too long as well.