d-ronin/dRonin #1240: UAVO filtering option "Only show existing" doesn't work

Treeed, 2016-08-25:
* Seen in version (Samsara): 
* On Operating System (Windows 7): 

Issue details:
In the system tab when using the button "View Options" (with an eye on it) and checking "Only show existing UAVOs" the GCS crashes. This happens no matter if a flight board is connected or not. 

  I tried to translate the below (error info from Windows) as accurate as possible from German:

>   Problemeventname:	APPCRASH
>   Applicationname:	drgcs.exe
>   Applicationversion:	0.0.0.0
>   Applicationtimestamp:	00030000
>   Errormodulename:	StackHash_0a9e
>   Errormoduleversion:	0.0.0.0
>   Errormoduletimestamp:	00000000
>   Exceptioncode:	c0000005
>   Exceptionoffset:	0000000e
>   Operatingsystemversion:	6.1.7601.2.1.0.256.1
>   Areascheme-ID:	1031
>   Additionalinformation 1:	0a9e
>   Additionalinformation 2:	0a9e372d3b4ad19135b953a78882e789
>   Additionalinformation 3:	0a9e
>   Additionalinformation 4:	0a9e372d3b4ad19135b953a78882e789

tracernz, 2016-08-25:
Thanks. Any chance you could try a newer build such as https://github.com/d-ronin/dRonin/pull/1205#issuecomment-242435498? Newer Windows builds include crash reporting so we can get an actual stack trace which we can process into this at the other end: http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgKCz4s4JDA :+1:

Treeed, 2016-08-25:
I tried downloading from jenkins, but my current internet connection seems to prohibit that. Maybe someone else can reproduce the issue, I don't think it's limited to me.

mlyle, 2016-08-25:
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgOCQvokKDA

PR #1242 (2016-08-27): uavobjectbrowser: don't crash on "only show existing"
Fixes #1240
commit 4c980d19 (2016-08-27): uavobjectbrowser: don't crash on "only show existing"

Fixes #1240
closed 2016-08-27 by tracernz