d-ronin/dRonin #1249: after auto tune stabilization page showing default pids

redparchel, 2016-08-28:
* Seen in version (release version or githash): Release-20160720.1: 20160723-74b7422c
* On flight controller (if applicable): naze32 rev6c

Issue details:
start on "stabilization" tab, connect to FC, go through prompted autotune wizard. after clicking "done" on the autotune wizard the stabilization page still shows the default pids (i believe it should display the new pids read from the board) i clicked save (out of habit) saving the defaults back to the board and undoing the autotune pids.

mlyle, 2016-09-01:
I can't reproduce this-- when I run autotune, my stabilization page appears to update immediately.

Was tested on a naze32 clone (quanum pico)

mlyle, 2016-09-05:
Closing because no-repro :/

closed 2016-09-05 by mlyle