d-ronin/dRonin #1252: RawHID: Crash Report (Disconnection)

tracernz, 2016-08-29:
http://dronin-autotown.appspot.com/at/crash/ahFzfmRyb25pbi1hdXRvdG93bnIWCxIJQ3Jhc2hEYXRhGICAgKCgtKcKDA

Looks like the connection was closing and we were in the process of destroying the write thread while it was still running, thus the mutex was a null pointer. The null pointer check in QMutexLocker must have been optimized away? http://code.qt.io/cgit/qt/qtbase.git/tree/src/corelib/thread/qmutex.h?h=v5.6.0#n126

P.S. Might be nice to add a feature allowing developer comments on crash reports to track this kind of thing instead of Github issues.

mlyle, 2016-08-31:
Could very easily be something -inside- qmutex being null or becoming null.

PTDreamer, 2016-12-06:
Wow that web interface is a killer! nice!

mlyle, 2017-01-26:
@PTDreamer Sorry I didn't notice that, but it's @tracernz's fine work--- based on your crashreporter stuff, of course.

This was fixed by #1519.

closed 2017-01-26 by mlyle