d-ronin/dRonin #1254: BrainRE1 watchdog reset

mluessi, 2016-08-29:
* Seen in version (release version or githash): 2016-07-20.1
* On flight controller (if applicable): BrainRE1

Issue details: Flight controller resets in-flight, message "BOOT: INDY WDOG"  shown on reboot (see video below). This has only be seen by very few users, so it could be a hardware defect (unlikely, I think), or something related to the particular setup where it occurs. Logging was enabled in the video below, but it also occurs without logging. 

https://www.youtube.com/watch?v=OBUXzPko5MA&feature=youtu.be

First thoughts on what could be causing this:

- Spurious interrupts to video driver / OSD
- BMI160 driver
- OSD code* 
- VTX Config code*

*not sure if/how an rtos task would trigger the watchdog. Any other ideas?

thatjoshguy, 2016-08-29:
That's my video above.  some additional notes:
 - This board was provided before the official release to review, not sure if it is different from final production run?
- logging was enabled for that flight in order to see what was causing the issue.  I have had similar issues since day one that I assumed were caused by RC failsafe, however...
- This flight was with a new FrSky XJT module and new D4R-II receiver connected via direct solder and running PPM into the BrainFPV RE1.
 - I believe autotune was still enabled during this flight, though not assigned to a flight mode.  I think I have had this happen with autotune completely disabled, but am unsure.
 - I had several flights before this with no issue, and a short flight after this without issue.  Seems to happen at least once every 4 or 5 flights (though I am usually much closer to the ground when it happens!)
 - I was able to pull a few logs, thoguh some would not complete the download and would cause the RE1 to lock up requiring a reboot.  Logs: https://www.dropbox.com/s/n2euxmbqf9cvjeu/logs.zip?dl=0
 - I have IR transponding disabled
 - I have VTX Config enabled and in use
 - I am running Oneshot (not sure if that matters - all four motors stop at the same time)

If there is any other data I can provide, please let me know. I love this FC but at the moment I cannot trust it.

mluessi, 2016-08-29:
@thatjoshguy thanks for the details. If it works for you, it may be best if you hold on to the RE1 you have until we have some ideas what could be causing this, so you could test code with a bugfix. If we can't figure it out, BrainFPV will send you a replacement and you can return the one you have for analysis.

thatjoshguy, 2016-08-29:
will do.  I'll get some pics of the board jsut in case something is damaged and see if I can get actual logs

mluessi, 2016-08-29:
@thatjoshguy can you export your config (File->Export UAV Settings) and post it somewhere?

thatjoshguy, 2016-08-29:
https://www.dropbox.com/sh/az9yyto498v6rvc/AAA5n1beRhMo7l_ZDTRLm1DNa?dl=0

Above link contains the one log that I was able to dump successfully (not sure if it has the data we need or not), my UAV settings file, and a log from the desktop software for what happens when i try to dump logs.  I can dump log 0 successfully, log 1 freezes consistently between the 450-650 sector mark.

thatjoshguy, 2016-09-03:
@mluessi flew today with no issues at all.  Best flight yet actually.  I did two things:

1. Turned Autotune completely off.  it wasn't on a flight mode, but it was still check-marked  in GCS

2. Disabled Logging - Now this is interesting because I only enabled logging the other day to see why this was happening.  I disabled it and saved settings, did some other stuff - but when i went to the System view in GCS where I can see every variable and looked in the LoggingSettings section it seemed it was still set to LogOnArm.  I set this to LogOff and saved/rebooted.  I thought this worked but when I checked just now this was set to LogOnArm again?  Is this expected behavior?  I'm not entirely sure the logging is actually disabled.

mlyle, 2016-09-09:
Only one manifestation, no longer repros, closing.

raetron, 2016-10-14:
I believe I am having this same issue. I am running the latest version dRonin on a BrainRE1. 
I have video and logs if needed.

tracernz, 2016-10-14:
> I have video and logs if needed.

Yes please. :+1:

mlyle, 2016-10-14:
Also if you have a configuration dump that would be helpful too.

mluessi, 2016-10-17:
One user on facebook reported the issue went away after disabling logging. Was that you @raetron? Maybe it's a CPU overload?

mluessi, 2016-11-03:
Several users have reported this on release 2016-10-04.1. Disabling logging seems to fix it, so there could be a bug in the logging module that somehow only affects RE1 or logging may be overloading the CPU.

LitterBugs, 2016-11-18:
Had one of these on my maiden FPV tonight with a brand new RE1.  Had done an autotune and one brief LOS flight before this flight. Logging was enabled.

DVR of FPV feed:  [https://www.youtube.com/watch?v=WN68a2EAvTI](url)

Export and Config from RE1 post incident
[LitterBugAwk210RebootExport.zip](https://github.com/d-ronin/dRonin/files/598959/LitterBugAwk210RebootExport.zip)

Log file from flight
[dRonin-2016-11-17_21-54-25.zip](https://github.com/d-ronin/dRonin/files/598969/dRonin-2016-11-17_21-54-25.zip)

Mosaic of frame grabs at the moment the problem happened
[image]

mlyle, 2016-11-18:
Basically these sort into two categories:

* "It resets when I flip / go fast / etc".  I've helped 3-4 people with these, and they've been power quality issues, either from defective PDB or from build issues.
* "It's related to logging".  I am not sure this is different from the other group.  I know of one person on facebook who had the issue fixed by disabling logging.

It's interesting that something happened to the video just before the reset.  Had the vsync glitched before anywhere in flight?  (Is this a coincidence, or did something happen with the sync separator circuit--- which is hardware but probably cares about board voltage-- at around the same time?)

mlyle, 2016-11-18:
@mluessi -- it sure looks like there's something wrong with RE1 hardware or with the support code for RE1.  How do you plan to proceed?

LitterBugs, 2016-11-18:
This was a BOOT: INDY WDOG.  Discussion on the Facebook channel suggested that ruled out power.

LitterBugs, 2016-11-18:
I see no indication anywhere else in the DVR feed of OSD sync issues.

mlyle, 2016-11-18:
@LitterBugs That's incorrect.  I can generate indy wdog by dipping power (but it has to be a short dip-- longer dips become brownout reset or power-on-reset).

LitterBugs, 2016-11-18:
If there is any more information you would like, I'll happily provide it.  I used the log viewer to review the log file and did not see anything obvious.  I would imagine the logging may not have been written out at the moment the error happened and caught what happened, but it is there for everyone to review.

mlyle, 2016-11-18:
@LitterBugs -- Yes-- you're correct-- log files are not very useful for hunting crashes and better for flight anomalies, etc.  Right now we don't even measure power quality in any way.

The sync glitch right before is interesting--- it doesn't completely rule out firmware, because an interrupt storm or similar could delay us dealing with the beginning of the frame.  But the fact that it happens for at least 3 OSD frames in a row and is offset exactly the same sure looks like something analog in the sync separator circuit.  As far as I know the only thing that could affect the sync separator circuit and firmware execution is power quality.

LitterBugs, 2016-11-18:
I added the power monitoring fields to the screen in the log viewer.  Looks like there is power monitoring in the log to me...
[image]

mlyle, 2016-11-18:
I'm saying we don't ADC the microcontroller's voltage etc to know if it's consistent.  (Can measure VCC indirectly-- the microcontroller has an internal 1.2V reference that we can measure what fraction of VCC it is-- but we don't do that right now).  Probably not a bad idea for a future enhancement.

LitterBugs, 2016-11-18:
Any testing I can do that doesn't involve flight? Really hesitant to put all this shiny new gear back in the air and risk damaging it, something, or somebody. Pretty fortunate that there was no visible damage from a freefall at that height. I've thought about securely strapping it down out in the yard and doing some static testing. Any other ideas?  I can attach a scope if there is any signal in particular that we should be looking at.

mluessi, 2016-11-18:
@LitterBugs one thing you could try is to disable logging and strap the quad down. If it doesn't reset, try flying it. If no resets happen then, it would indicate that the logging module is the causing the problem.

@mlyle I will investigate. An interrupt storm of the video sync could definitely cause this. We actually had this problem with the OG Brain and added some code to prevent it from happening (the same code is also there for RE1). I will try and see if I can cause resets by having spurious video sync detections.

mlyle, 2016-11-18:
@mluessi one problem with the logging module theory is some boards -- with
fairly diverse configuratilns-- have dozens of flight hours with logging.
E.g. @ufodziner, mine.-- both external serial and flash logging.  Other
people's boards seem to be prone to resets.  One guy has said turning
logging off has cured it.  Other people have had resets tied to aggressive
maneuvers.

I am not blaming a video sync interrupt storm.  It is just interesting
vsymc gets messed up-- offset late-- immediately before the reset for
several frames.  Since this is timer triggered on a high priority interrupt
path, it seems like some common (non-software?) factor is affecting sync
and the reset.

LitterBugs, 2016-11-18:
I'm going to post a slo-mo starting several seconds before it happened when I get home.  There was a brief drop in amperage for several frames just before it all happened too.   Here's a different plot of the log.  Added Throttle, CPU load, and Watchdog status to the default fields.
[image]

LitterBugs, 2016-11-18:
Would hooking up an external OpenLager instead of or in addition to using the internal flash memory be of any use?

mlyle, 2016-11-18:
Can you tell based on timing... whether the last frames with the vsync
anomaly are in the time period in the log?  Perhaps the battery current
field more be most illustrative.

mlyle, 2016-11-18:
(Because if vsync looks mixed up, but everything looks happy still from the
point of view of the flight log... that's interesting).

@mluessi-- it might be worth teaching the RE1 firmware how to measure the
3V3 rail.

mluessi, 2016-11-18:
OK, I'm pretty sure it's video sync related. Did 3 tests using settings from @LitterBugs :

1. Logging, no video connected. No reset in 30min
2. Logging, video connected. No reset in 30min
3. Logging, video connected with frequent video GND disconnects to generate spurious interrupts. Resets frequently.

I assume this can be fixed in SW. Will look into it further.

mlyle, 2016-11-18:
Interesting.  I wonder if this will also fix some of the "resets on aggressive acro" people-- maybe conducted noise is responsible for some of this.  Thanks @mluessi.

LitterBugs, 2016-11-18:
In the video when the clock shows 2:41-2:42, there is a current drop to 6.6A then back up to 41A that does not show in the log.  Looks like the log stops just before I dropped throttle and raised it back up.

mluessi, 2016-11-18:
Actually, I just did one more test: 3 but with logging disabled, doesn't reset. Maybe the increased CPU utilization by logging together with the spurious interrupts is causing it.

LitterBugs, 2016-11-18:
So is that noise on the video input causing it, or from elsewhere?

mluessi, 2016-11-18:
@mlyle that's definitely possible. Motor noise and ground shift during aggressive maneuvers can cause sync problems

LitterBugs, 2016-11-18:
Could the camera adjusting to changing light be contributing to this?  Notice how the brightness changes about the time it happens too in the mosaic of frames.

LitterBugs, 2016-11-18:
On radical moves, you could be getting radical changes in video levels.

mluessi, 2016-11-18:
@LitterBugs if you hold your camera towards a lamp and cover/uncover it with a finger, do you see the OSD jumping / tearing? If so, it can definitely cause the problem. I can reproduce it here by enabling logging and changing the sync threshold.

LitterBugs, 2016-11-18:
Let me go take my props off, arm it with logging enabled and try moving between light and dark rooms......

mluessi, 2016-11-18:
Using your finger to cover/uncover the camera is the easiest way to test if the sync threshold is correct. If the OSD jumps/loses sync, you need to adjust it.

LitterBugs, 2016-11-18:
Doing the hand trick will sometimes make the screen jump.   Hard to predict when.  Maybe fire up the strobe light to really freak it out. ;)

mluessi, 2016-11-18:
@LitterBugs change the video sync threshold so it doesn't jump. This should "solve" the problem with the current firmware. Naturally, it shouldn't cause the reset and I'm looking into fixing it.

LitterBugs, 2016-11-18:
FYI...  Much easier to trigger in a darker room......

LitterBugs, 2016-11-18:
Hmmm.  Actually I was bumping the antenna which was causing the RF link signal to jump....  Need to do this direct connected to my ground station monitor.

LitterBugs, 2016-11-18:
Just triggered a reboot by adjusting it the wrong way...  I think we are on to something here.

LitterBugs, 2016-11-18:
Got a SW reboot message not the INDY WDOG.....

LitterBugs, 2016-11-18:
Original 120
Lowest 113
Highest 122

New value 117

I think the reboot I got was triggered by me hitting the reboot button in the configurator.

LitterBugs, 2016-11-21:
Martin,
My baro arrived today.  Looks like weather should permit me to get a pack or two in on the way home from work tomorrow.  Logging is turned off, Vsync tuned to 117, barometer on.

Cheera!
Karl

mluessi, 2016-11-22:
I did some debugging this morning. It seems like a UsageFault occurs during the VSYNC ISR. By default, this fault is not handled, so it will hang in `_unhandled_exception` until the watchdog resets the CPU. 

The CFSR is 0x20000, which indicates and "INVSTATE: Invalid state UsageFault". The stacked PC is 0x0, which is suspicious. From what I have read, this can be caused by a stack overflow. Have tried increasing `__main_stack_size__`, `__process_stack_size__` and `UAVO_CB_STACK_SIZE` with no change. Any other ideas? Is the ISR stack part of the main stack?

tracernz, 2016-11-22:
@mluessi What's the value of the usage fault status register? It's a half word at 0xE000ED2A. See http://infocenter.arm.com/help/topic/com.arm.doc.dui0553a/Cihcfefj.html#Cihgbdbi for further details.

LitterBugs, 2016-11-23:
Got out to the field tonight at dusk and flew two packs.  No problems at all.  Locked in and really pleased with the way it flew.

blowback, 2016-11-23:
I have just wired in a replacement RE1 & mPB (many thanks!) and can exhibit this behaviour on the bench. I've done the video sync calibration (min=118, max=124, actual=121). On the bench, lens cap on, just arming the motors is enough to trigger an INDY WDOG.

There is weirdness when it comes back up: instead of the usual 5 ESC beeps, I only get the first 4. You get the final beep when you disarm again. Often at this stage only one of the motor/ESC channels is working, or they're all working but one of them is "graunchy". 

There is an awful lot of noise on the video when the ESCs are doing their thing. My VTx is at the back. The signal goes over the top of RE1, and the power goes underneath mPB (maybe that's the problem?)

I've just discovered that if I disconnect the camera completely, the system seems a lot more stable. Maybe even stable enough for flying - will verify tomorrow.

Here's a couple of vids I made of what's going on:

https://www.youtube.com/watch?v=sYorJjeJXnk
https://www.youtube.com/watch?v=SW_6cRUXx0A

mlyle, 2016-11-24:
@blowback -- do you have logging enabled, or are you able to manifest this crash without logging?

blowback, 2016-11-24:
Sorry, those were both with logging disabled (vanilla Dronin install on new RE1 board), camera connected, lens cap on.

This morning I have just flown two packs without a single issue (with the camera electrically disconnected).

tracernz, 2016-11-24:
> Sorry, those were both with logging disabled (vanilla Dronin install on new RE1 board), camera connected, lens cap on.

That's fine. Proves it's unrelated to logging or the extra CPU load it generates. 👍

gpthefabman, 2016-11-29:
Hey guys, I'm running into the exact same problem. Has it been resolved? If so, please let me know. I'm completely new to Drones and flying. This has been a steep learning curve and the loss of power mid flight has been aggravating. I cant wait until I can actually start learning on this thing and not just fixing it due to mid flight power losses. 

Any help would be greatly appreciated. I'm excited to join the community!

Thanks

Re1 mPB
aikon esc 30a
tiger f40vII

mluessi, 2016-11-30:
OK, found the problem: `PIOS_Vsync_ISR` calls `dev_cfg->set_bw_levels`, which for RE1 then accesses the FPGA via SPI. When logging is active, this seems to be causing a conflict, as the SPI is also used for logging etc. Spurious video sync interrupts will cause it to call the function repeatedly (as the video system detected switches back and forth).

I propose to move the function call into the OSD task instead of calling it from the ISR. Thoughts?

mlyle, 2016-11-30:
@mluessi  --- that sounds great.  Yes, if you try and claim a SPI in an ISR, then you can block on the semaphore and blocking in an ISR is really bad.

Thanks for running this down.

PR #1473 (2016-12-02): Brain/BrainRE1 OSD: set PAL/NTSC settings from OSD task not ISR
Fixes #1254 

When the video system changed between PAL and NTSC, the ISR was setting the black and white levels. For BrainRE1, this resulted in a SPI transfer to the FPGA from the ISR, which could lock up when the SPI was in use (e.g. for logging). This PR moves the calls to the OSD task and cleans up some things. Same changes also applied for Brain for consistency.

Status: Bench tested for RE1 (could not reproduce problem anymore, even with 250Hz logging and a lot of spurious interrupts). Needs test on Brain
closed 2016-12-02 by tracernz