d-ronin/dRonin #1266: GCS: need to make it obvious not to use non-delay arming modes

mlyle, 2016-08-31:
This is a big source of support overhead and user frustration.  Needs a fix for Quixote.

tracernz, 2016-08-31:
Is it feasible to remove them? I guess they could be useful for ground vehicles.

mluessi, 2016-08-31:
I don't see a harm in removing them. For a ground vehicle, always armed could be an option and delay shouldn't hurt either.

mlyle, 2016-09-01:
Non-delay modes are useful for aircraft which are not likely to be static during arming-- e.g. perhaps planes.  But I think the tab can be adjusted be be nicer.

tracernz, 2016-09-01:
We could also consider hiding the non-delay options for multis.

mlyle, 2016-09-01:
Here's what I'm thinking of:

!(http://i.imgur.com/xs4qpkv.png)[http://i.imgur.com/xs4qpkv.png]

tracernz, 2016-09-01:
I think the "Arming Settings" box is better but still much to confusing... I'll whip something up when I get home tonight to see what you think. I like the "Disarming Settings" box, especially the separate failsafe and idle timeouts. I'd change the text for the idle timeout to "Automatically disarm if throttle is low for:" though.

mlyle, 2016-09-01:
Connects to #447

mlyle, 2016-09-01:
While I'm sympathetic / hope we can get this perfect.. I'm happy with anything easy to implement that reduces the rate of end-users running into this problem by, say, 50%.

PR #1279 (2016-09-26): GCS: New Arming Settings Tab
@mlyle This is what I have to date. Spent the bulk of my time tonight on back-end stuff (in configtaskwidget, the warnings stuff etc.) so the front-end will be nicer and easier.

TODO:
* Make "Your vehicle is armed by" and "throttle must be low" checkboxes actually do something (others are plumbed. Note that the throttle one is read-only for stick arming modes, might be best to just hide it and add "throttle low and" in front of the options).
* Plumb the checkboxes to disable/zero stuff
* Fill the above items from the UAVO values
* Add more warnings

How it looks s [... cut]
closed 2016-09-26 by mlyle