d-ronin/dRonin #1284: opengl init not working on windows software accelerated env

mlyle, 2016-09-03:
reported by @jhitesma 

[image]

tracernz, 2016-09-03:
Unfortunately can't repro this running under RDP (RDP requires ANGLE/GLES).

mlyle, 2016-09-22:
No repro, but not ready to close either.

jhitesma, 2016-09-22:
Any benefit to me trying a more recent build?  I assume nothing has changed but haven't tried to repo with any newer builds in a few weeks.

mlyle, 2016-09-22:
Nothing has changed.

How much RAM do you have?  I wonder if that is really an out-of-memory error.

jhitesma, 2016-09-22:
That machine only has 4G, and it's hurting.  I could try with everything else shut down to see if it works then.

mlyle, 2016-09-22:
Would be helpful to see if it cures it-- then we could easily close this.

jhitesma, 2016-09-24:
Just tried it with everything else shut down (even chrome's background stuff) and same error.

Watching Taskmanger my memory usage never went past 56% 
So doesn't look like a memory issue. 

The video card on this computer is old enough that there aren't official drivers for it for W10 so I'm not surprised it's having issues related to video.

tracernz, 2017-01-31:
Playing with different Qt OpenGL backends a bit... GLC_Lib doesn't play nice with any except the software one. Will be good to get rid of it soon (WIP) and then we can probably fix this quite easily.

mlyle, 2017-02-01:
Still broked-- too bad

tracernz, 2017-02-01:
#1540 should be interesting.

tracernz, 2017-02-03:
This was fixed for @jhitesma by https://github.com/d-ronin/dRonin/pull/1540

mlyle, 2018-02-04:
Not seen in 5ever

closed 2018-02-04 by mlyle