d-ronin/dRonin #1288: Vehilce Setup Wizard resets items set by Radio Setup Wizard

yamahabest, 2016-09-03:
* Seen in version (release version or githash): http://jar.lyle.org:8080/job/dronin/577
* On flight controller (if applicable): AQ32

Issue details:

When you first run the Radio Setup Wizard, and then the Vehicle Setup Wizard, at RC Input the Throttle, Roll, Pitch, Yaw and FlightMode settings, at Flight Mode Switch Settings and Arming Settings are all reset.

I did it in that order, and skipped the Radio Wizard option at the end of the Vehicle Setup Wizard (did think to myself, I did already setup that). Was expecting to be able to fly. No joy, after connecting back to the GCS, I noticed those settings changed.

Could be that I was doing it in the wrong order.
Might confuse new users also.

tracernz, 2016-09-03:
Intentional, because you select a new receiver type in the wizard. The radio wizard is supposed to run AFTER the setup wizard, hence it offers to do that. https://github.com/d-ronin/dRonin/blob/next/ground/gcs/src/plugins/setupwizard/vehicleconfigurationhelper.cpp#L364

yamahabest, 2016-09-03:
Agree, should do all channels then, yes.

PR #1298 (2016-09-06): setupwizard: program all channels
* Don't just program a few chosen ones.  Fixes #1288
* Program Min/Neut/Max with appropriate values for protocol
commit 9c34788d (2016-09-06): setupwizard: program all channels

* Don't just program a few chosen ones.  Fixes #1288
* Program Min/Neut/Max with appropriate values for protocol
closed 2016-09-06 by tracernz