d-ronin/dRonin #1291: DSM sat bind issue

jhitesma, 2016-09-03:
* Seen in version (release version or githash):  Build #575 e009c5e
* On flight controller (if applicable): airbot acro F4

Issue details:

Using a DSM2 sat on this airbot F4 board.  sat is pre-bound to my TX.  If I power up the board by connecting my battery I get a fast blink from the sat and no bind.  But if I power up off USB I get either a slow blink or a solid light (as expeted depending on if I've reset the TX or not.) and it's bound.

Kind of a major issue since it means I'd have to drag along a computer to fly at the field as just powering up off a bat I get no bind.

jhitesma, 2016-09-06:
@mlyle - do you thing this affects all targets?  I can try moving the DSM sat to my sparky, sparky2, Naze32 or Brain (original) and see if it does the same there there if that will help at all.  (though doing it on the sparky2 and Brian would be a bigger pain since those are in vehicles currently.)

mlyle, 2016-09-06:
It would be helpful if you could try this satellite on another board, yes.  Is it a genuine Horizon satellite or orangerx, etc?

My best guess is this is an electrical issue where it thinks it's getting the binding pulse on startup, even though we're not sending it (and indeed, this port is not *capable* of sending the binding pulse if the inverter is present).

mlyle, 2016-09-06:
[image]

This is the input structure -- so it's indeed different from other targets.

jhitesma, 2016-09-06:
It's a lemonRX sat.  I'll try it on some other boards tonight and see how it goes.

jhitesma, 2016-09-07:
Just a point of reference, litterbug (the guy with the owl frame in the forum) says he's using a Lemon DSM sat on his RE1 and not experiencing this issue.  But he's on samsara so that doesn't rule out a regression since then.  I didn't have time to swap the RX around last night and probably won't tonight either but will try to make time later this week to confirm it's a board issue and not a code issue.

mlyle, 2016-09-07:
No chance of a regression.  (no related change).

It's either hardware (the satellite, the airbot board) or something unique to Revolution.

jhitesma, 2016-09-07:
Ok.  The sat worked fine on my naze32 with dRonin and litterbug is using a lemonRX sat on a BrainRE1 with no issues.  So I don't think it's an issue with the lemon sats (though mine is a DSM2 and his is a DSMX so I can't say 100% for sure...but this is the only sat I have so I don't have hardware to swap to rule it out.)  

I also don't have a regular revo to compare with so I can't really help rule that out.

So I'm not sure there's anything else I can do to help at this point unless I get another sat or a regular revo.  I was thinking about maybe picking up one of the $20 acro revo's but not sure if that would help or just potentially cause more confusion.

LitterBugs, 2016-09-08:
FYI, I am on GitHub but LitterBug was taken so I am LitterBugs here.

mlyle, 2016-09-08:
I don't think we need to go and do an inventory of all targets to see if binding works.  The main question is whether there's an issue with the revo target causing satellites to go into binding mode at startup.

It'd be the bootloader / how pins float.  I really, really suspect the input structure here, though, because there's a 10k pullup from the sbus input structure.  If the DSM satellites have a similarly valued pulldown to prevent false pulses and assume whoever's sending has a ~50k impedance microcontroller pullup at max, bad things will happen.  This is also something that may vary between satellites-- the guys designing lemonrx may have different input structure themselves (pullup value, ESD prot, etc) and different firmware for what they consider a binding pulse.

mlyle, 2017-01-13:
I think this has been confirmed to be particular to these Airbot AIO targets

closed 2017-01-13 by mlyle