d-ronin/dRonin #1293: Airbot F4 vehicle wizard puts DSM sat on wrong port

jhitesma, 2016-09-03:
* Seen in version (release version or githash): Build #575 e009c5e
* On flight controller (if applicable): airbot acro F4

Issue details:
Using vehicle setup wizard all went well, except the sat was configured on the flex port by default and I had to change it to main port for this board.

mlyle, 2016-09-04:
worst part of this is having to change the wiring diagram :(

tracernz, 2016-09-06:
@mlyle We had this but you removed it. :stuck_out_tongue: https://github.com/d-ronin/dRonin/commit/c46418bf496f76bf4322e3fdfc9173ea82bed78b

mlyle, 2016-09-06:
Yes, except you can't bind.  that's an issue too :/

tracernz, 2016-09-06:
Let's not make things worse for real Revo then. :wink:

jhitesma, 2016-09-06:
Since these boards are already in production and being sold it sounds like our best bet is to just have an "errata" type of page in the wiki about this.  I'd say maybe a warning in the GCS, but I doubt anyone would read it and I'm not sure if we could identify these boards from the other revoish boards.

If the OSD boards aren't in production yet and have this same wiring maybe we can ask airbot about revising them to match real revo to minimize the number of goofy boards.

tracernz, 2016-09-06:
If we detect the different flash (we already do this), we can identify these boards, and set the board revision right? Then we could do some relatively basic trickery in GCS to make it work for this board (albeit with no bind support).

mlyle, 2016-09-06:
Unfortunately the different flash is both on "revo acro" and these boards.

jhitesma, 2016-09-06:
Could detect the presence of the OSD maybe?  Is the version with OSD going to be a new target?  Don't want to wander too far off topic, just wondering if maybe "different flash without OSD" would be a good way to identify this board.

mlyle, 2016-09-06:
The "revo acro" has different flash.
The "Airbot F4 without OSD" has different flash and the DSM satellite port.
The "Airbot F4 with OSD" has different flash and the DSM satellite port.

We can't tell the difference between "revo acro" and "Airbot F4 without OSD".

mlyle, 2016-09-08:
I think this is WONTFIX, because real revo should be best supported.

closed 2016-09-08 by mlyle