d-ronin/dRonin #1301: Autotown calculates yaw values whilst the GCS isn't

yamahabest, 2016-09-05:
* Seen in version (release version or githash): http://jar.lyle.org:8080/job/dronin/589/
* On flight controller (if applicable): AQ32

Issue details:

The GCS says it can't calculate the yaw values, however autotown does.
Are they using different calculations? 

I am not sure if it's actually an issue, or by design.

mlyle, 2016-09-05:
Autotown's threshold for refusing to calculate yaw values is different, like I said on IRC.

Autotown is a nice to have / developer tool.  It doesn't need to exactly match.

closed 2016-09-05 by mlyle

yamahabest, 2016-09-06:
Fair enough, but are the values from autotown usable, or should I stay far from them?

mlyle, 2016-09-06:
The output is currently the same, but the conditions on which they are displayed is different.