d-ronin/dRonin #1310: GCS: should remove logging playback

mlyle, 2016-09-07:
1. It's not useful.
2. It can't play telemetry logs.
3. It confuses users.

I had a quick go at it but ended up shelving it because there's a lot of interdependency with GCS logfile writing-- will have to start from scratch/more carefully.

tracernz, 2016-09-07:
Can you describe what it actually does (what logs it DOES play, what it was designed to do), and what you want it to do? Hard to grasp why this is a bug from your description. 😛

mlyle, 2016-09-07:
GCS has a log playback feature.   If you give GCS a log that it recorded, it can "play it back" and pretend to be receiving this data for the first time-- so the graphs widgets will slowly scroll the data in realtime, you can see the data in the PFD.

It will only work for logs recorded by the GCS over telemetry sessions.  I have never been able to use it for anything useful, and it seems like a user shows up confused by it about once a week.

tracernz, 2016-09-07:
Yes, I thought that. It can be useful for GCS development as you can reproduce things over and over repeatably. Perhaps we should hide it from regular users (expert mode checkbox can handle this).

mlyle, 2016-09-07:
it's really, really messily implemented :/  but if you use it...

tracernz, 2016-09-07:
Heh, if it ain't broke... :stuck_out_tongue: If you really want to remove it I won't stop you. :wink:

mluessi, 2016-09-07:
AFAIK, it doesn't playback logs from flash because the time stamps are different. It shouldn't be terribly hard to fix this and it may be useful for navigation flights etc

tracernz, 2016-09-07:
Flight logs won't have the GCS telemetry state though right? So GCS won't behave properly/think a board is connected?

mlyle, 2016-09-24:
If people use it, i will kill this bug.

closed 2016-09-24 by mlyle