d-ronin/dRonin #1312: Autotune: consider running gyro filter on incoming points

mlyle, 2016-09-08:
Looking at the Librepilot import of our autotune, they run the configured gyro cutoff lowpass filter on the points as they come in.

This is marginally correct to do-- there is about 1.5ms of group delay @50Hz with a 55Hz cutoff filter.  But at the same time, we have appropriate fudge factors that a 1.5ms offset is not that big of a deal and by softening the edge we probably add dispersion to the measured tau.

BUT-- it seems from recent facebook discussion that one cause of failure of tunes to converge is vibration.  This can cause beta to run away super high and the magnitude of the wiggle to go to zero or nearly zero.  This would help address that.

glowtape, 2016-09-09:
Gonna test this tomorrow.

(Altho I don't have any significant vibrations in my current quad.)

glowtape, 2016-09-10:
I guess it's gonna take some more time. For some reason the patch linked above blows up and causes a boot loop. And I'm not seeing the forest for the trees.

glowtape, 2016-09-12:
This makes quite a difference on my frame and it doesn't have anything considerable in vibrations. I've had yet to get a sub-20ms tau tune with triblade props. Adding the gyro filter to it shaved like 3ms tau (I'm at 18ms now), increased gains on roll and pitch, and for the first time ever I have EKF noise that's just one digit in front of the comma _on all axes_.

This is a typical triblade tune (it's my saved tune of the ExpoM build, which was done in no wind):
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAoPOvigkM

This is the tune I've just done with gyro filtering (also no wind):
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAoMvemwkM

The higher gains and lower tau aren't causing any issues, the quad flies mightly fine, so it didn't spit out bullshit values.

glowtape, 2016-09-12:
I just tried my Q5040 props and filtered the tau and gains for better.

That said, they are dinged up since I don't have new ones, which seemed to have an effect on all of it, too. They tuned 17-18ms before, now the unfiltered tune (which I did back to back with the filtered one) resulted in 22ms. Gains of the dinged props are also lower than seen in older tunes.

Quadblades unfiltered:
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICA4JjH7AgM

Quadblades filtered:
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAoKC-pAoM

glowtape, 2016-09-12:
Put some DAL 5045BNV2 on it. Situation is only partly the same. Tau didn't budge but EKF noise is lower and gains are higher.

Twinblades unfiltered:
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAoM2A-QgM

Twinblades filtered:
http://dronin-autotown.appspot.com/at/tune/ahFzfmRyb25pbi1hdXRvdG93bnIYCxILVHVuZVJlc3VsdHMYgICAoM2A-QoM

glowtape, 2016-09-12:
Yeah, I just redid an unfiltered autotune on the tris to verify, tau ended up the same as filtered, but gains are still higher, noise still lower.

Despite the higher gains, there didn't appear any bad effects, like oscillations/ringing.

mlyle, 2017-03-16:
Obviated by #1599, which doesn't do exactly this, but does something very good to the same effect.

PR #1599 (2017-03-16): sensors: Implement N-th order gyro and accel lowpass filter.
Alternative version of #1595, that puts the filtering in sensors and applies to both gyros and accelerometers. Works on the bench, would be interesting to see how it affects current autotune, but weather and all.

Still need to fix the naming in GPL headers, need to figure out the proper group names.

Might be an idea to include the magnetometer?

fixes #1312
closed 2017-03-16 by mlyle