d-ronin/dRonin #1320: RE1 bootloader-- how does it work?

mlyle, 2016-09-08:
Doesn't it use the '405 linker scripts/memory map/startup code?  Are we just getting incredibly lucky that it doesn't fail?

tracernz, 2016-09-09:
F4 linker scripts are messy too. BL uses a different set to BU/FW.

tracernz, 2016-09-11:
It works because BL doesn't use CCM RAM I guess? And it only uses the beginning of the flash. Pretty dodgy.

mluessi, 2016-09-13:
Other than the linker script, what is wrong? The clocks get set up by `cmsis_system.c` in the board info dir. Are there other files that should be different for RE1 than for F405 targets?

mlyle, 2016-09-13:
I don't know.  The linker script/memory layout is definitely wrong at minimum.  Are register maps for all peripherals we touch right, etc?

mluessi, 2016-09-13:
The register maps should be correct; they are selected by defining `STM32F446xx` vs `STM32F40_41xxx` for F405 targets. This is done correctly in the BL and the BU makefile. I will look into selecting the correct linker file for the BL/BU.

PR #1356 (2016-09-15): STM32F446: fix startup and linker code
Fixes and cleans up linker scripts for BrainRE1 and adds global linker scripts for STM32F446, also adds additional vectors to the startup code.

Fixes #1320
closed 2016-09-15 by mlyle