d-ronin/dRonin #1331: OSX Release Builds currently do not work

mlyle, 2016-09-09:
```
Process:               dRonin-GCS [56645]
Path:                  /Users/USER/Desktop/*/dRonin GCS.app/Contents/MacOS/dRonin-GCS
Identifier:            org.dronin.gcs
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           dRonin-GCS [56645]
User ID:               238135

Date/Time:             2016-09-09 12:55:50.013 -0700
OS Version:            Mac OS X 10.11.6 (15G31)
Report Version:        11
Anonymous UUID:        61497D56-BD3F-F614-1B44-4A3037CE86D2

Sleep/Wake UUID:       B13F616F-3543-49FD-978C-7010E8983ADE

Time Awake Since Boot: 880000 seconds
Time Since Wake:       13000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /Users/USER/Desktop/*/dRonin GCS.app/Contents/MacOS/dRonin-GCS
  Reason: image not found

Binary Images:
    0x7fff68d56000 -     0x7fff68d8da47  dyld (360.22) <DC81CC9D-651A-3A45-8809-928282052BD3> /usr/lib/dyld
    0x7fff99556000 -     0x7fff999ccfff  com.apple.CoreFoundation (6.9 - 1258.1) <943A1383-DA6A-3DC0-ABCD-D9AEB3D0D34D> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
```

mlyle, 2016-09-09:
```
          cmd LC_RPATH
      cmdsize 136
         path /Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/tools/Qt5.6.1/5.6/clang_64/lib (offset 12)
```

mlyle, 2016-09-09:
Need to check #1203, et al

tracernz, 2016-09-09:
Can't repro with http://jar.lyle.org:8080/job/dronin/606/

tracernz, 2016-09-09:
With the above linked artifacts: 
```
          cmd LC_RPATH
      cmdsize 40
         path @loader_path/../Frameworks (offset 12)
```

tracernz, 2016-09-09:
Okay, I looked through backlog in IRC and found it is the [artifacts from #1250](http://jar.lyle.org:8080/job/dronin/584/) causing this issue. I can indeed repro with that build.
```
          cmd LC_RPATH
      cmdsize 136
         path /Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/tools/Qt5.6.1/5.6/clang_64/lib (offset 12)
```

So it's something in that PR? The offending artifacts are older than last night's nightly which doesn't have this issue.

tracernz, 2016-09-09:
Tried the [nightly built immediately after #1250 artifacts](http://jar.lyle.org:8080/job/dronin/585/Nodes=osx/), found the same issue:
```
          cmd LC_RPATH
      cmdsize 136
         path /Users/Shared/Jenkins/Library/Developer/org.jenkins-ci.slave.jnlp/workspace/dronin/Nodes/osx/tools/Qt5.6.1/5.6/clang_64/lib (offset 12)
```

tracernz, 2016-09-09:
Don't see it in http://jar.lyle.org:8080/job/dronin/589/Nodes=osx/. Something fixed it between job 585 and 589. 😕

tracernz, 2016-09-09:
http://jar.lyle.org:8080/job/dronin/588/ is the first build after job 585 that doesn't have this issue.

tracernz, 2016-09-09:
New artifacts for #1250 are fine: http://jar.lyle.org:8080/job/dronin/611/Nodes=osx/

mlyle, 2016-09-10:
I'm going to reboot the OSX builder.  Backgrounds have stopped appearing on .dmg

mlyle, 2016-09-11:
[image]

Now the backgrounds are back, but the applications shortcut isn't there and things are out of place.

mlyle, 2016-09-11:
Artifacts work OK though.

PR #1338 (2016-09-11): macosx package: bump base size of dmg
The space only gets used if it's filled.  Assumed that this fixes #1331.
commit cd4da22d (2016-09-11): macosx package: bump base size of dmg

The space only gets used if it's filled.  Assumed that this fixes #1331.
closed 2016-09-11 by tracernz