d-ronin/dRonin #1332: max7456 support on aq32 and revolution

mlyle, 2016-09-09:
Currently in mpl-7456 branch

cc @dustin 

* [ ] Get the targets to turn it on and off appropriately in hw settings
* [ ] Confirm playuavosd still works
* [ ] Download fonts to '7456.
* [ ] Make the OSD widgets do actual OSD-ish things

dustin, 2016-09-09:
What's special about AQ32?

mlyle, 2016-09-09:
AQ32 has an external SPI "OSD" port that the users have hooked max7456 boards to in the past.  If we can secure testing (@yamahabest ? ) it would be nice to support this.

jihlein, 2016-09-09:
The AQ32 external SPI port was used to drive this:

https://www.sparkfun.com/products/retired/9168

Not sure if anything like this is still available or not.

mlyle, 2016-09-09:
Yup, on AQ32 it'd be purely for existing users (and to get additional early OSD help/feedback).

yamahabest, 2016-09-10:
Sure, I can test this. I just have to plug it in again.

yamahabest, 2016-09-10:
http://www.hobbytronics.co.uk/max7456-osd-breakout

yamahabest, 2016-09-10:
Whilst this is nice, why is this going to be supported?
With those MinimOSDs and the PlayUAV OSD support on its way, are there still people interested in this?

mlyle, 2016-09-10:
@yamahabest there are flight controllers coming out with '7456 on the board, as stated above, which require no additional components for OSD if we add software support.

yamahabest, 2016-09-10:
Ah, sounds good, thanks.

mlyle, 2016-09-10:
We also talked about it on IRC before, I think... saying that I'd support it because of those boards... :P

yamahabest, 2016-09-10:
Ehhh.. Could be, yes. Let me know when I can test something.

PR #1345 (2016-09-29): Direct MAX7456 support - for AQ32 and future F4 targets
I WIP my hair back and forth.

Fixes #1332 

Large portions derived from https://github.com/UncleRus/MultiOSD

# General TODO

* [x] Doxygen for PIOS_MAX7456
* [x] Suck in and use more of the existing Max7456 driver 
* [x] PIOS_MAX7456 wait-for-vsync hack
* ~~Vsync interrupt support [defer, I think]~~
* [x] Meaningful content for widgets (@dustin is working on this some)
* [x] Font built into image
* [x] Paths cleanup
* [x] Font loading
* [x] Font selection (just have a couple built-ins) (@dustin)
* [x] Font tweaking
* [x] Splash / reset  [... cut]
closed 2016-09-29 by tracernz