d-ronin/dRonin #1333: Wishlist:  Ability to load UAVO sets in Android app without installing new app

jhitesma, 2016-09-10:
Honestly not even sure if this is possible.  I do like that @mlyle added multiple UAVO sets to the new test build of the app which is a definite step in the right direction.  But if there's any way we could allow the app to say grab UAVO sets from the cloud it sure would make it a lot more usable.   

But I'm probably not a very typical use case since I'm so often running development/next builds.  Not sure if UAVO sets are modular enough that this would be possible or not but if it's at all possible it sure would be nice to have.

mlyle, 2016-09-10:
Almost certainly wontfix with how it works.

mlyle, 2016-09-10:
(may be prohibited by android license agreement to pull down code like this dynamically)

closed 2016-09-10 by mlyle