d-ronin/dRonin #1337: Obviously Bogus GPS Value

dustin, 2016-09-11:
I noticed this when reviewing some flight footage.  I had four frames of this, surrounded by an estimate of about 321m from home:

[image]

This plane's pretty slow, so anything that'd warp me like that would be obviously wrong.

dustin, 2016-09-11:
Doh, that was the frame immediately before the bad GPS.  Here's one of the bad frames:

[image]

dustin, 2016-09-11:
```
        <object id="0xE2DBC224" name="INSSettings">
            <field name="AccelVar" values="0.00300000003,0.00300000003,0.00300000003"/>
            <field name="GyroVar" values="9.99999975e-06,9.99999975e-06,9.99999975e-05"/>
            <field name="MagVar" values="10,10,100"/>
            <field name="GpsVar" values="0.00100000005,0.00999999978,0.5"/>
            <field name="BaroVar" values="0.00999999978"/>
            <field name="MagBiasNullingRate" values="0"/>
            <field name="ComputeGyroBias" values="FALSE"/>
        </object>
        <object id="0xBA09F7C2" name="StateEstimation">
            <field name="AttitudeFilter" values="Complementary"/>
            <field name="NavigationFilter" values="Raw"/>
        </object>
```

mlyle, 2016-09-11:
I think this is expected/WONTFIX/indicates a wiring problem.  Curious if you get lots of checksum errors on the GPS.

It's important the actual navigation modules do appropriate glitch filtering (and we may not have enough).  But raw shouldn't for sure, and it's arguable OSD should display the actual raw data even if we know it's garbage.

dustin, 2016-09-11:
It's reasonable for the OSD to display raw GPS, for sure, but stats, e.g., should ignore teleportations.

[image]

tofly1990, 2016-09-12:
HI, my fixedwing always bow after tuen around,I check the attitude ,when it turn around the attitude is not righjt,I think this is the motion acceleration effects?

dustin, 2016-09-12:
@tofly1990 I'm not sure what you mean.  Are you saying your GPS produces incorrect values when you turn around?

My most recent flight only had a couple of GPS outages and on specific hard turns, so that makes sense in my case.  But they were just loss of signal, not incorrect information.

tofly1990, 2016-09-13:
@dustin sorry my english is poor.
 sparky2, no gps, fixedwing, complementary;
 after turn around,the attitude always distortion and the fixedwing head will Dive to the ground.

dustin, 2016-09-13:
@tofly1990 this bug is "Obviously Bogus GPS Value" -- if you don't have a GPS, it's probably not related to your problem.  :)

You may find more help in [the forum](https://forum.dronin.org/forum/).

tofly1990, 2016-09-13:
@dustin  ok!

mlyle, 2018-04-19:
So this is really interesting.

1. the lat and long are really close, e.g. not a trivial flipping of bits from true position
2. UBX protocol has checksumming...
3. 536939800 = 0x20010D18, 536682500 = 0x1FFD2004  They're very close to each other in value, but not in bit values.
4. NOGPS?