d-ronin/dRonin #1349: dRonin logo doesn't play nicely with a black Windows task bar

yamahabest, 2016-09-12:
* Seen in version (release version or githash): -
* On flight controller (if applicable): -

Issue details:
The letters are "invisible".
I believe mlyle already has some updated icons, which include some white lining around the logo.

tracernz, 2016-09-18:
Which icon are we using, because branding/gcs.ico looks like it should be fine and I vaguely remember seeing it on Windows. Must be grabbing the transparent PNGs from somewhere for the Windows 7+ taskbar icon?

jhitesma, 2016-09-19:
You mean a black title bar?  Because a black (or at least dark grey....but doesn't look like it would matter) taskbar looks ok to me:

[image]

Black title bar though does look pretty crummy:
[image]

yamahabest, 2016-09-25:
Ehm.. Yes, that also.
However, I should have taken a screenshot of it happening, as I can't seem to reproduce it anymore on the task bar. It looked the same as your screenshot of the titlebar. Letters dissapeared against the black background, only the orange triangles visible.

yamahabest, 2016-09-25:
Reproduced: https://snag.gy/OaIkvY.jpg

mlyle, 2016-10-07:
Any idea how you reproduced it?  Any chance that machine just has an outdated icon that isn't getting replaced by the install?

PR #1412 (2016-10-11): gcs: attempt to fix title bar appearance on windows
When the user uses a dark taskbar.

Fixes #1349.
closed 2016-10-11 by tracernz