d-ronin/dRonin #1355: Flight Stats Enhancements

dustin, 2016-09-14:
I'm wanting to make a more useful stats screen.  I'm not sure we collect all the most important stats.  The main things I want to know:

* How much of my battery did I consume?
* How long did I fly?
* How far did I go?
* How fast did I go?
* How high did I go?
* What's the maximum current I drew?

The first three are sort of accounting while the last three are limits.

The major thing missing from our current stats is current stats (what's the max current drawn).

What else do people want out of their game over screen?

For comparison.

Brain Stats Screen:
[image]

MWOSD Stats Screen:
[image]

tracernz, 2016-09-14:
Max filtered current draw, or max spiky current draw? 😛

dustin, 2016-09-14:
Well, somewhat filtered.  At the very least, max likely to have been displayed at any point.  Maybe we can plot the current usage pdf.

ufoDziner, 2016-09-14:
I'd like the max roll/pitch/yaw rates prior to my crash, or whatever it was that made them excessive.  Personally the distance/speed/alt stuff doesn't apply to me at this time.  Lowest battery voltage (checking for excessive sag), and the amp draw as @tracernz mentioned.

dustin, 2016-09-14:
Yeah, I'd left out the rates because of the crash thing.  That's interesting.  Rotation rates, but not in the last few seconds.  Given that, it should be OK to display the numbers, but perhaps change how they get into stats.

Re: distance/speed/alt -- omitting stuff we don't have sensors for should be a given.

mluessi, 2016-09-14:
Things that don't have a sensor should already not be displayed. There could be a bug, but I have never seen distance etc. on a quad without GPS / nav filter configured.

dustin, 2016-09-14:
@mluessi This isn't so much complaining about what's being displayed, but trying to figure out what could be better.  I always miss the max current thing as well as flight time.

I think the current OSD code is probably doing the right thing with what it's got, but I'd like it to have more/different things for my tastes.