d-ronin/dRonin #1360: Unable to connect after first time to a powered FC

yamahabest, 2016-09-16:
* Seen in version (release version or githash): Development
* On flight controller (if applicable): AQ32

Issue details:
I am not sure what is causing this.

Steps I am going through:
1) Power the FC by battery.
2) Connect USB cable to laptop which is already running GCS.
3) FC and GCS connect just fine.
4) Disconnect USB cable from laptop, keep battery connected.
5) Reconnect USB cable to laptop.
4) FC and GCS are unable to connect, I see only the first two bars of RX blinking for a while. After which nothing is shown anymore.

Same procedure without a battery connected work flawlessly.

dustin, 2016-09-16:
I think you may be the only user of this board.  There have only been 21 distinct boards ever having been seen (and we did have a bug a while back that caused a small number to be double-counted).

Does this happen for you on anything else?  It seems like it'd be a lot easier to drop support for AQ32 than to fix it.  (e.g. I wouldn't even know where to get one if I wanted to work on it)

jihlein, 2016-09-16:
I use this board a lot.  I think the observed issue is due to the lack of USB vsense and a long bootloader timeout.  I see the observed behavior too,  and just ignore it.   I know of at least 3 users.  Not a large number, but users never the less.

closed 2016-09-16 by mlyle

yamahabest, 2016-09-16:
Ok, @jihlein, thanks for the explanation.

I figured the issue could be across more boards, but I guess not.
For the rest the board works great, have two of them flying quads.

I would like it if the target is not dropped, but I can only hope.

mlyle, 2016-09-16:
I don't want to drop aq32.  But it will always get somewhat worse support, because I don't have one.

tracernz, 2016-09-16:
Likewise, I have some PCBs but they were going to cost too much to populate.