d-ronin/dRonin #1363: Drop support for F1

mlyle, 2016-09-16:
For formal discussion here (and mentioning it on IRC, and it can make its way to the forum/broader community after we gather initial thoughts on it.)

We're nearing the end of Quixote, and it looks like it makes F1 a little better and fixes some regression/annoyance there.  It also shows us how limited F1 is, and it's annoying to worry quite so much about RAM, etc.

I'd like to do one more release after Quixote that supports F1, and declare the release "Long-term support"-- we'll fix critical bugs in it for some time even after subsequent releases ship, and maybe even backport trivial features.

After that, nix it.  (Except F1-based pipxtreme/oplink/taulinkmodule and F0-based flyingpio).

Thoughts?

mlyle, 2016-09-17:
Some discussion about this on IRC earlier today.  Summary of thoughts:

* Some people think 2 more releases with F1 is too aggressive of a deprecation schedule.
* There was discussion about how to announce this properly for minimal impact to user population-- specifically we should hit on
  * There are cheap alternatives that do work with us much better.
  * The reasons why (constant tax on development, needing to do many pieces of work twice, etc)
  * That there will be some degree of long-term support for F1
* There was some skepticism (probably rightful) on how many features will get backported to LTS
*  Maybe we need to support USB DFU for better experience with some of the more common first-party produced FC's (DTFc, Lux)

tracernz, 2016-09-17:
I agree with DFU, but not really related to F1 deprecation, since half our F1 targets are much worse off in that regard.

mlyle, 2016-09-17:
Yah.  the DFU is in the context of-- if we're going to ask a thousand people to move off CC3D, what can we do to make things easier for them.

mluessi, 2016-09-17:
I think it's a good plan to drop F1 and have a last LTS release for it. Also DFU would be nice for things like LUX etc that come without the dRonin bootloader.

jhitesma, 2016-09-19:
I think most of those comments from IRC were mine ;)  At least they echo my concerns.

I'm generally in favor of dropping support for F1 boards, but I think we need to be a little less aggressive on the timetable due to our stats currently showing over 60% of logged boards being F1 based.

A LTS release with backported issues is nice in theory, but I suspect that in practice as soon as we drop support for F1 there will be a flurry of changes that will make backporting impractical for most issues that get fixed.  So we'd either end up with a LTS release that isn't really supported - or which just makes more work.  Neither of which is a great solution.

But it is pretty much the only practical solution, which is why I think extending the timetable one more release makes sense.  If we could get a notice into Quixote at this point that warns when an F1 board is connected that support will be phased out soon...then @mlyle's proposed timetable is more palatable to me.  Then in post-Quixote we could have a more comprehensive message with suggested alternatives listed/linked and that could be the last release to support F1.  

Even then, a cautious side of me would still want to extend it one more release just due to the popularity and prevalence of F1 boards.  

Either way I think DFU support would be a big win.  But agree with @tracernz that it's more of a separate issue.

PR #1608 (2017-03-10): Remove support for F1 flight controllers (CC3D, Naze32)
Has the time come?  Fixes #1363.

There's a pleasant amount of negative diff here, though the 'acro' attitude code needs to get pulled also (with accordant changes to Lux/DTFc and eliminating the altitude-runaway-if-no-baro bug.  Since that affects more targets I think it's a good consideration for a subsequent PR.
closed 2017-03-10 by tracernz