d-ronin/dRonin #1364: CPU warning with OpenLager

ufoDziner, 2016-09-17:
* Seen in version (release version or githash): d5615b2f
* On flight controller (if applicable): Brain RE1

Issue details:
Logging max rate at 100 gives a constant "CPU" warning on the OSD.  Disabling logging on the Mainport removes the warning.

mlyle, 2016-09-17:
Can you attach an associated log?

ufoDziner, 2016-09-17:
Still can't attach a zip.  So, here's the link.  https://dl.dropboxusercontent.com/u/59924814/LOG018.zip

mlyle, 2016-09-17:
```
UAVO_TaskInfo(name=UAVO_TaskInfo, time=73.68, uavo_id=0xFECD7750, StackRemaining=(
200, 592, 1592, 400, 184, 0, 304, 0, 128, 0, 0, 88, 0, 0, 0, 0, 0, 0, 0, 0, 0, 400
...
1), FALSE(0), TRUE(1), FALSE(0), FALSE(0), TRUE(1), FALSE(0)), RunningTime=(0, 15,
 22, 10, 0, 0, 0, 0, 4, 0, 0, 15, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 
0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0))
```

Corresponds to 15% actuator, 22% attitude, 10% sensors, 4% manualcontrol, 15% stabilization, 1% onscreen display.

I don't know, maybe RE1 just can't handle high-rate logging.

mlyle, 2016-09-17:
It's likely the dma-for-logging stuff might help this.  Seems to be brain specific though, because I log at 500Hz on Sparky2 with only 75%-ish CPU use.  You're around 81% which is in the warning threshold but not really a problem as I've said before.

closed 2016-09-17 by mlyle