d-ronin/dRonin #1365: Offer PID value selection for common airframes

mluessi, 2016-09-17:
We currently have autotune, which can give pretty good results. However, unfortunately one of the most frequent problems user have is failed autotunes. Likely, this is caused by vibrations of the airframe and can sometimes be fixed by soft-mounting, replacing motors, etc. Future improvements to AT should also reduce this problem to some degree.

However, it is still possible that AT doesn't work for some reason, or that a user uses an airframe type that is not supported by AT (e.g. a airplane). I think it would be nice to give users options to use default PID parameters that should work OK for their airframe and will only require a small amount of manual tuning. Maybe we could have something that is connected to the cloud, so we can add airframes between releases.

Note that I'm not saying that we should abandon AT, but one problem we currently have is that many potential users equate dRonin with AT, so it they don't want to use AT they don't want to use dRonin. I think having a selection of good defaults will help many users getting in the air more quickly and provide overall a better user experience for them.

Thoughts?

mlyle, 2016-09-17:
I think the emphasis should be on making autotune work and not adding too much complexity to the UI.  I'm not opposed to this in principle, but autotune is our main differentiator and it's important that we make it work well for people who have come to do AT.

tracernz, 2016-09-18:
Dup. of #1988 ?

mlyle, 2016-09-19:
This is the approach that the LP guys have taken.  The big downside is that so much of this comes down to props and ESC response.

I am wondering (and we can prove out) whether deadbands would let us get away with somewhat higher default gains / default I especially.

jhitesma, 2016-09-19:
I kind of like the suggestion peabody had in that tau issue which ties into a few things I know @dustin and I have discussed in IRC about the ability to save/restore tunes from the cloud - which doesn't sound that hard to implement.  Kind of solves a number of issues if implemented in a way that allows both privacy and the ability to share tunes (which is admittedly the tricky bit.)

I'm not a big fan of "just fly the defaults" or "pick something close from a list" - that just seems to go against the things that attracted me to Tau/dRonin in the first place.  

At the same time some of our current defaults aren't the greatest.  We've already increased the rate defaults which is nice, and the expoM stuff is killer.  I think I also saw a PR about increasing the default gyro rate on some boards which is long overdue IMHO.  I have no issue with our default PID's...but I'm not building super high end no holds barred racing setups.  Though I also suspect most people aren't doing that either and those seem to be the users who have the biggest issues with the defaults (and should in a perfect world be the most capable of overriding them intelligently...not that that seems to be the case in reality.)

robert-hoffmann, 2016-11-24:
I like this idea too.

Don't forget that most noobs (me included) just buy ARF/RTF/BNF quads. Having a list of good PIDS for these should not pose a problem for the (multiple esc/motor) hardware variations.

What kiss tried here
http://kiss.flyduino.net/dwkb/user-shared-pids/
Seems way too simple to me..

You guys already have autotown, but the main problem is, it's just a big list, and you have no clue as to if the tune is good or bad.

..maybe some kind of option to rate the tune. Like after applying the tune, and actually having flown it, i come back to dronin, and give it a rating, so that afterwards the data is at least useful (other than just for debugging)

mlyle, 2018-03-16:
No

closed 2018-03-16 by mlyle