d-ronin/dRonin #1367: Sparky2 PPM wizard setup issue

inis1981, 2016-09-18:
* Seen in version (release version or githash): 
Based on Qt 5.5.1 (32 bit)
Built on Jul 24 2016 at 03:29:07
From revision 74b7422c 20160723
UAVO hash c9177491
* On flight controller (if applicable): Sparky2.0 (Hobbyking)

Issue details:
RCVR PPM doesn't work if setup through wizard, bypass wizard it works fine. Setup manually then go through wizard PPM will no longer work. Only fix I've found is erasing the board and flashing it/start over. Does this on three Sparky2.0's I have.

mlyle, 2016-09-18:
Does this happen if you set channels 7 and 8 to 0 min/neutral/max and the associated bank to 50Hz?  

If that fixes it, fixed in #1317 (Vehicle setup wizard no longer configures channels you're not using).  There's a hardware bug in sparky2 where the same timer is used for PPM decode and channels 7 & 8.

mlyle, 2016-09-18:
(Believed to be dup #1317)

inis1981, 2016-09-18:
mlyle I went through and changed motors 5-6, 7-8, and 9-10 to 50, then min/max for motors 5+ to 0 us. All good now.

mlyle, 2016-09-18:
Cool, thank you for confirming.

closed 2016-09-18 by mlyle