d-ronin/dRonin #1373: PiOS:HAL: Wrong baudrate for DSM

tracernz, 2016-09-20:
Correct baudrate is 125k according to [Spektrum protocol spec](http://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf). We use 115.2k.

jhitesma, 2016-09-20:
Looks like openlrsng is wrong as well:  https://github.com/openLRSng/openLRSng/blob/master/RX.h#L688

And CF:
https://github.com/cleanflight/cleanflight/blob/1ea7608ac2802752fb545b98ba8f76f9b82b9db3/src/main/rx/spektrum.c#L51

Probably everyone :D  Wonder who got it wrong first....

DTFUHF, 2016-09-20:
> Looks like openlrsng is wrong as well: https://github.com/openLRSng/
> openLRSng/blob/master/RX.h#L688
>
> And CF:
> https://github.com/cleanflight/cleanflight/blob/
> 1ea7608ac2802752fb545b98ba8f76f9b82b9db3/src/main/rx/spektrum.c#L51
>
> Probably everyone :D Wonder who got it wrong first....

Maybe Spektrum got it wrong, as all of those projects predate that document.

tracernz, 2016-09-20:
@DTFUHF The doc specifically mentions 115.2k implementations. :wink: 

@SpektrumFPV, can you provide any information on the SPM4648? It's advertised as supporting SRXL but I found no mention of this in the user manual. There is some talk around that it may feature a revised DSM protocol too?

SpektrumFPV, 2016-09-20:
@tracernz It doesn't support SRXL, just the remote receiver/sat protocol. I brought this up with somewhat to have that fixed on the website. Sorry for the confusion, and thanks for bringing it to my attention!

As for the baud rate specs in the doc, you are correct. I'll quote the section of it for anybody with any doubts.
"The packet is transmitted at 125000bps, 8 bits, No parity, 1 stop (8N1).
For those UARTs which are not capable of that speed, they will also work at the more-standard
115200bps. "

tracernz, 2016-09-20:
This code to "detect the resolution" satellites bound in external mode is very sketchy and depends entirely on configuration of channel order by the looks: https://github.com/d-ronin/dRonin/blob/next/flight/PiOS/Common/pios_dsm.c#L198-L206. That would explain why some people have problems with it.

tracernz, 2016-09-20:
@SpektrumFPV Thanks. The SPM4648 also says it supports 20 channels. How does this work with the protocol?

mlyle, 2016-09-20:
an 8.5% error in baud scaler is a big deal (125kbps vs 115200).  Plus, presumably many spektrum clones choose 115200.

We probably need to pick a compromise value like 120000 :P

tracernz, 2016-09-21:
> Plus, presumably many spektrum clones choose 115200.

May, or do? Receiving a signal from the transmitter seems to be the bigger battle for most of the DSM clones in my experience. :stuck_out_tongue: .

mlyle, 2016-09-21:
Yah, would be nice to measure.  Even if they do-- targeting 120000 is better than targeting 115200 if Spektrum is sending us 125kbps.

tracernz, 2016-09-21:
So, uh, I hooked up a genuine Spektrum SPM9645, and it definitely speaks at 115200, 125000 is all framing errors...

http://imgur.com/a/fEGWE

tracernz, 2016-09-29:
I don't believe the Spektrum document is accurate based on my measurements of a genuine device.

closed 2016-09-29 by tracernz