d-ronin/dRonin #1377: Bluetooth low energy telemetry connection doesn't work on OSX

mlyle, 2016-09-21:
Because OS X (like iOS) leaves it up to applications to pair with BLE devices.

The QT bluetooth API would allow us to talk to HM-10A etc.

mlyle, 2017-03-10:
This doesn't affect me anymore, and I don't think anyone else is screaming for it.

tracernz, 2017-03-10:
Agree with closing this. I'm interested in working on this for iOS at some future date but not likely to be anytime soon.

Anything you do with BTLE is going to be manufacturer and/or chip specific since the BTLE standard doesn't define profiles (e.g. Serial Port Profile in original BT spec). The single characteristic that HM-10 uses for serial is documented on various blogs etc. so it at least should be relatively straightforward.

tracernz, 2017-03-10:
Bringing this one back. 😉 
[image]

mlyle, 2017-03-12:
PR pls? :P

tracernz, 2017-09-09:
Rebased this recently, kind of works on mac but it seems to not have enough throughput for the session to come up gracefully (the update-on-change uavos dominate everything else even before the session is established, eww). Crashes on Linux. https://github.com/tracernz/dronin/commits/mrc-blueteeth