d-ronin/dRonin #1378: DSM Binding needs to error if configured and unavailable

tracernz, 2016-09-21:
Took me a while to get one bound. Sparky1 is too slow, Sparky2 is too slow when powered by USB, and silently fails to even attempt it when on the RxPort. We definitely need to give feedback (i.e. a configuration error) for the latter case!

mlyle, 2016-09-21:
Why is Sparky1 too slow?  I've done it before.  (For a long time had a sparky1 quasi-dedicated to this).

tracernz, 2016-09-21:
Probably the powered by USB bit (didn't state that first), it took 2 seconds before if sent the binding pulses (sparky2 was 1 second, requirement is 200 ms). We should do something about that because it's actually most likely you'll want to bind while setting up with GCS via USB.

tracernz, 2016-09-21:
Could use a backup register or two to do Spek binding in the bootloader. 
We currently use the first 4 backup registers, total on each chip is:
* F103: 10x 16-bit registers
* F303: 16x 32-bit registers
* F405/446: 20x 32-bit registers + 4 kb backup SRAM

I envisage something like:
* Hit bind button in GCS
* Backup register is set with index to an array of the bind pin configs
* Prompt to unplug board, then re-plug
* Bootloader starts and sees the bind backup register is set, sets up bind pin and sends binding pulses, then continues as normal

mlyle, 2016-09-21:
> Prompt to unplug board, then re-plug

Don't those registers need power to vbat to be persisted?  Though there is some eeprom emulation on these parts I think.

One option for non-DTFc/Lux/BrainRE1 would be to watch for a battery power edge in the firmware and send the binding pulse immediately.

tracernz, 2016-09-21:
Duh, of course.

mlyle, 2016-09-21:
Oh, sorry, these are 3V3/always get receiver power.  For non-3V3 receivers you need one (on non-DTFc/Lux/BrainRE1).

tracernz, 2016-09-21:
If only we could switch the 3V3 supply to the Rx port.

LitterBugs, 2017-03-20:
Is there still any active development going on around DSM binding?   I have several targets to test with.  Several targets will successfully put the RX into bind mode, but will not complete bind with TX.  Think the issue may be startup delay related.  Can scope with triggering to get timing and pulse counts.

mlyle, 2017-03-20:
If you can get them into bind mode, but bind doesn't complete-- you're meting tmiing, but the number of pulses is wrong.  Note you may need to subtract one or two pulses from what's expected.

tracernz, 2017-03-20:
Newer Spektrum receivers can be bound over UART so it'd be nice to have that but I don't have any new enough hardware currently (and I doubt many do).

LitterBugs, 2017-03-20:
RMRC has the new telemetry RXes in stock. [https://www.readymaderc.com/store/index.php?main_page=product_info&products_id=5931](url)  I've thought about picking one up for LOS telemetry.  Usually the OSD have everything I need.

LitterBugs, 2017-03-22:
I picked up a Spektrum SPM9645 satellite and one of the SPM4649 Telemetry units to test with.  Been using the LemonRX LM0037 and LM0007 and it is possible they have a tighter timing window.  I've also been using the Spektrum SPM4648 with autobind that does not need bind pulses.  The SPM4649 telemetry RX needs higher voltage (ie 5V) and can also use a bind plug.