d-ronin/dRonin #1380: Contact

joshuabardwell, 2016-09-22:
@digitalentity's message to me asking me to contact him via email disappeared from the raceflight issue shortly after he posted it. I'm not sure if you tried to take me up on my offer and the response was deleted.

If you want to talk one-on-one about the RF GPL violation, email me joshuabardwell@gmail.com.

mlyle, 2016-09-22:
Opening issues on other projects is not an appropriate way to get in contact IMO.   The contributors to this project don't need emails and stuff popping up in their bug tracker, etc.

I'm not really interested in litigating this in the sphere of public opinion.  RF is violating the license, and if it continues, there will likely be legal consequences.

mlyle, 2016-09-22:
P.S. it seems that post is still there.. however, they are deleting anything that Boris posts.

closed 2016-09-22 by mlyle