d-ronin/dRonin #1383: Windows build/packaging issues

jihlein, 2016-09-23:
* Seen in version (release version or githash):  Latest next
* On flight controller (if applicable): N/A

Issue details:  After the move to MSVC, building on my Windows 7 laptop gives a couple of errors I've been unable to track down.  I've removed the complete environment, and started from a clean slate, still get the same errors.  The errors are thrown from the latest next branch and my own development branches.

1)"GCS_BUILD_CONF=release FLIGHT_BUILD_CONF=release make -j7 all"  This command will completely build all bootloaders, updaters, firmwares, entire flash loads, and GCS.  GCS works fine, and the firmware is flyable.  But it throws an error at the end I've not seen before: "Cannot add commands to previously defined target FORCE".  Here's  a screen dump of the last few output lines of the build:

[image]

2)After the above command completes, executing "make package_flight" terminates with this error:

[image]

And sure enough, the folder C:\dronin\build\package-20160923-316f75d3 is empty, the zip file does not exist.

FWIW, before a lot of the build changes went in, I was able to successfully run:

GCS_BUILD_CONF=release FLIGHT_BUILD_CONF=release make -j7 all make package_flight
make package_all_compress
make package_installer

and generate a working installer.  I've been unsuccessful trying to find out the exact PR(s) that are causing the issue.

mlyle, 2016-09-23:
It doesn't look like the information as to what's going on is in those screenshots.

You can redirect the build logs to a file with `>whatever.txt 2>&1` and then attach them to this issue. .. also it would be helpful if you provided non-parallel versions of the output (I know this takes forever).

mlyle, 2016-09-23:
Just trying to guess from the last line of output..  is that a createprocess line for zip failing that you've cut the top off of?  . I think it's trying to run zip?  have you run make zip_install / is the zip tool present?

jihlein, 2016-09-23:
Non parallel build cooking, will attach when complete.  Zip tool is present, but re ran make zip_install for grins before I started the non parallel build.

mlyle, 2016-09-23:
I think it's complaining it can't run zip-- will be interesting to see.

jihlein, 2016-09-24:
It looks like the rebuild of the zip tool  did the trick, I can successfully make install packages now.  Can't test them at the moment, so will leave this open a little longer.

jihlein, 2016-09-24:
Installer works, closing.  Thanks @mlyle.

closed 2016-09-24 by jihlein

mlyle, 2016-09-24:
I'm super glad this is working for you now.  I bet when the tools were broken you ended up with a broken zip install.   Ugh.  I am not a windows fan.