d-ronin/dRonin #1384: GCS: Session management sometimes gets stuck in handshake

tracernz, 2016-09-24:
On the GCS side:
[image]

Weird that the flight side is showing 0 data rate too... Objects are being received just fine and the CPU load is 32% so not pegged.

mluessi, 2016-09-24:
I think this is a duplicate of https://github.com/TauLabs/TauLabs/issues/1418

jpbarraca, 2016-11-01:
I think I'm having a related problem when using a Sparky2 in MacOS with Quixote.
The FC reports data, sensors are received and scopes update, but:
- It never "gets link"
- The GCSTelemetryStats object is updating with strange values: Very unstable RXDataRate (highs and lows or 0), TXFailures keeps increasing. Status is Connected.
- In linux (ubuntu 16.04, Virtualbox) gcs seems to operate correctly

mlyle, 2017-06-28:
I think I plan to close this in favor of #1571, as it's the core obvious repro case.

closed 2017-06-28 by mlyle

tracernz, 2017-06-28:
It's a different issue (USB on flight side vs. session management handshake on ? side for this one)?

mlyle, 2017-06-28:
I think/thought session management is correct, and this glitch happens when USB falls apart (e.g. it's OS/version dependent, seems to relate to hotplug, etc).  That is, this seems to be a more general description of  #1571 --- am I wrong?

tracernz, 2017-06-28:
No this is when we're talking and sending objects back and forth but the session handshake gets "stuck".

mlyle, 2017-06-28:
OK, I've not seen that.  Can it be repro'd?

tracernz, 2017-06-28:
I've not find a reliable repro (I have tried a little bit), just seen it happen very occasionally.