d-ronin/dRonin #1398: Receiver Type and Mode on RE1 image

ufoDziner, 2016-10-04:
* Seen in version (release version or githash): Quixote
* On flight controller (if applicable): RE1

Issue details:
  Win 10, Hardware tab, Receiver Type and Mode are on the image.

[image]

mluessi, 2016-10-04:
@ufoDziner what's your screen size? I can see that when I make the window quite small on my screen.

mlyle, 2016-10-04:
@mluessi you should drop a scroll widget in (and move messages up to be more visible)

ufoDziner, 2016-10-04:
@mluessi 11" 1366x768

mluessi, 2016-10-04:
@ufoDziner please test #1399 once the artifacts are built. I was only able to test it on Linux

ufoDziner, 2016-10-05:
@mluessi Works great!  Thanks!

PR #1399 (2016-10-05): GCS: Fix RE1 Hw config layout
Fixes #1398
closed 2016-10-05 by mlyle