d-ronin/dRonin #1405: No control after arming with logging >100

ufoDziner, 2016-10-06:
* Seen in version (release version or githash): 02f3e353
* On flight controller (if applicable): DTFc

Issue details:
  While testing #1388, I enabled logging to collect more data.  Settings were to Log on armed, Full Bore, 100.  After arming I had no throttle control.  Motors just idled then stopped after ~15sec.  I recycled power, and tried again.  This time the the motors didn't stop for a very long time (>1min), and I still had no control.

I then bench tested (propless) and noticed that after arming there was a CPU warning and the board rebooted.  Then I set the logging to 50, and it appeared to work on the bench.  I did not flight test.

jpbarraca, 2016-11-30:
Don't know if it is related: In a Sparky2, Log on armed, Full Bore, 50, resulted in a corrupted log. Flight was OK.

[dRonin-2016-11-30_22-03-59.drlog.zip](https://github.com/d-ronin/dRonin/files/623150/dRonin-2016-11-30_22-03-59.drlog.zip)

mlyle, 2018-02-04:
I cannot repro this

mlyle, 2018-04-18:
Attempted, still no repro.  Time to close this out.

closed 2018-04-18 by mlyle