d-ronin/dRonin #1416: Frsky smart port  doesn't work on X4R-SB

zipray, 2016-10-08:
dRoninGCS Release-20161004: 20160930-dd27a7f8): 
sparky1
Issue details:
[http://openrcforums.com/forum/viewtopic.php?f=5&t=9253&sid=fd4f0a6e4366cbc7a714d3a814d3aaf8](url)

mlyle, 2016-10-08:
It looks like you're trying to use some Arduino-encoding / overloading of the values which is not the way we stuff them into s.port.  I don't think we'd be changing our values, because many people are using lua scripts that assume the current values.  Does the firmware package you're running on the radio need to be modified to accept the values in our format?

zipray, 2016-10-08:
Thanks mlyle,I have a Ardunio mavlink encoder,when I use the APM FC through it access XSR  receiver,remote telemetry information display is normal. if I let Sparky FC output mavlink protocol, and then through the Ardunio mavlink encoder access XSR receiver is also all showed normal .The problem is when I used The Sparky FC output Frsky S.PORT protocol to access XSR receiver , there will be display content is not complete.I don't know Frsky standard format t1 and t2 respectively represent what,It is look like Mavlink to frsky using t2 transfer number of GPS satellites.

mlyle, 2016-10-08:
@zipray ---

FrSKY has a bunch of sensors that they sell for various purposes.

The Arduino encoder you have encodes things in a particular way.  It pretends some things are other sensor values.  Your RX is configured to assume that the fields are in that format chosen by the developers of the Arduino mavlink sport encoder.

Those are different than the interpretation of fields that FrSky has.  They are also different from the extensions/interpretation of fields that we have.

The way to fix this is either--- 1) move to running some of the lua scripts that have been written to interpret dR S.Port data.  2) adjust your RX-side configuration to interpret things the way that dR sends.  or 3) contribute code to dR to send the s.port data in the form that you'd like.  (No one here uses those formats/conventions and would not be inclined to write this, I think).

mlyle, 2016-10-08:
P.S. For any followup, please consider the forum instead of filing a bug report against dRonin.  Thanks!

closed 2016-10-08 by mlyle

zipray, 2016-10-08:
I see, thank you for your patience。